micropython/micropython #1500: RFC: Get rid of C stdio in unix port

pfalcon, 2015-10-11:
I was a bit surprised to see this line: https://github.com/micropython/micropython/blob/master/unix/mpconfigport.h#L194 . I thought we're staying close to real POSIX OS, and few places that have printf() are essentially conveniences for debugging of startup code. Well, turns out that basic platform output defined in terms of extra C stdio layer, and trying to compile without <stdio.h> shows that bunch of other places depend on it.

So, I propose to immediately get rid and ban usage of fwrite() (replace with write()), and look towards getting rid of other C stdio functions. Reasons:

1. Code bloat.
2. Extra dependencies for porting (fullish C library required, instead of pretty minimal subset).
3. Issues and funkiness in vendor implementations. For example, in framework-started android code, I see printf() not working (write(1, ...) works as expected).

dpgeorge, 2015-10-11:
+1

A few things that need to be dealt with:

1. py/mpconfig.h defines MP_PLAT_PRINT_STRN in terms of printf: https://github.com/micropython/micropython/blob/master/py/mpconfig.h#L799 .  This is for convenience and compatibility with behaviour previous to revamping the print system to use mp_print everywhere.  We could just not give a default for this and force ports to think about how they provide one (since it's a pretty fundamental thing, where stdout goes).

2. 64-bit float printing requires snprintf: https://github.com/micropython/micropython/blob/master/py/formatfloat.c#L409 .  This is not strictly to do with C stdio but is a dependency that needs to be removed.

pfalcon, 2015-10-11:
> py/mpconfig.h defines MP_PLAT_PRINT_STRN in terms of printf. We could just not give a default for this and force ports to think about how they provide one (since it's a pretty fundamental thing, where stdout goes).

Why not mp_hal_stdout_tx_strn_cooked?

> 64-bit float printing requires snprintf

Right, so dealing with floats is the place where we depend on libc/libm - simply because we can't know which float format a particular platform may use. (And then we have own implementation for single-precision IEEE floats, and it (ultimately) should be possible to easily switch witch float support to use - libc's or internal - for any port).

dpgeorge, 2015-10-12:
> Why not mp_hal_stdout_tx_strn_cooked?

Good idea.  In fact all ports define it to this.

dpgeorge, 2015-10-12:
Changing fwrite to write in unix means that debugging output from printf (eg showbc output) is now out of order with respect to normal print output of scripts (since script printing now goes directly to stdout, while debugging is buffered).

pfalcon, 2015-10-13:
> Changing fwrite to write in unix means that debugging output from printf (eg showbc output) is now out of order with respect to normal print output of scripts (since script printing now goes directly to stdout, while debugging is buffered).

What's the easiest way to reproduce it, so we stay on the same line. E.g. would `./micropython -v -v -c "print(1)"` work?

And the obvious way to address that would be to move debugging output to FD too...

dpgeorge, 2015-10-13:
One example of a change is tests/basics/exception_chain.py.  The warning is currently printed using fwrite and appears after the "print(...)", when in fact it really should appear before (and does when fwrite is fixed to write).

This might even be related to #1445.

Well, there is only a issue for debugging, which is not that big an issue for me.  Maybe we just make the change (ie fwrite -> write) and fix debugging things when they become an actual problem.

pfalcon, 2015-10-14:
If it's ok with you, I'd prefer this way too - to do it gradually, the end aim is to get rid of stdio completely anyway. So, if you have this change, please commit it.

dpgeorge, 2015-10-14:
Ok, done (see above linked commit).

I had to disable 2 cmdline tests because now their printing is messed up, due to showbc using buffered printf.

Also, I think we can eventually get rid of mp_plat_print and just use mp_sys_stdout_print everywhere.  No real reason to have 2 separate print channels.

pfalcon, 2015-10-15:
> I had to disable 2 cmdline tests because now their printing is messed up, due to showbc using buffered printf.

Ok, so maybe keep this issue open for some time, so these weren't forgotten (and to motivate doing the rest of many steps to resolve this completely).

 > No real reason to have 2 separate print channels.

Well, there's sys.stdout can be overridden by user, while diagnostic prints shouldn't be. Btw, I believe I worked on making sys.stdout overridable (on unix port) - simply because tests regularly want to do that, but it doesn't work in mainline...

pfalcon, 2015-11-13:
Next step: https://github.com/micropython/micropython/commit/3a6b3d230c60c3acfb212cec2bd80305232af42c

Native enabling of cmd_showbc after that still doesn't work.

pfalcon, 2015-11-16:
Another step: https://github.com/micropython/micropython/commit/79c4ec1102d1749ba2e05780158457fd27b0fcf9

pfalcon, 2015-11-16:
> Get rid of C stdio in unix port

But it's not possible to get rid of stdio in any more or less sizable app. That's because:

~~~
FILE *fp = fopen("/etc/resolv.conf", "r")
~~~

That code is in 3 of 3 "embedded" libc's I looked at  (uclibc, musl, bionic).

So, there will be a long way to unbloated static unix build...

pfalcon, 2015-11-16:
Ok, and we want be able to exterminate all printf's from unix build (e.g. because of 3rd-party components). That's old news, and stmhal deals with that from a long ago. That's how lib/utils/printf.c from https://github.com/micropython/micropython/issues/1553 pops up here. So, that file doesn't implement printf and friends - it *overrides* libc's printf and friends, and redirects them to uPy's mp_printf.

So, I propose to rename that file to printf_override.c and get it ready to be used in unix port.

dpgeorge, 2015-11-17:
> That code is in 3 of 3 "embedded" libc's I looked at (uclibc, musl, bionic).

Could always provide our own (unbloated) definition of fopen etc in terms of POSIX :)

> So, I propose to rename that file to printf_override.c and get it ready to be used in unix port.

I don't see the distinction here between overriding and providing an implementation.  Why not just make it lib/libc/printf.c?  And then change all calls to mp_hal_stdout_xxx to use mp_plat_print and associated functions?

pfalcon, 2015-11-17:
> Could always provide our own (unbloated) definition of fopen etc in terms of POSIX :)

I don't know why I skipped that idea, but I decided to just port lwIP's resolver to BSD sockets (pretty much done now).

> I don't see the distinction here between overriding and providing an implementation. Why not just make it lib/libc/printf.c? And then change all calls to mp_hal_stdout_xxx to use mp_plat_print and associated functions?

Again, the original idea of lib/libc/ was that it is standalone implementations of libc functions (i.e. something reusable, usually reused by us from other projects). utils/printf.c is clearly different from this.

 > And then change all calls to mp_hal_stdout_xxx to use mp_plat_print and associated functions?

Also not sure about this. I started with reluctantly converting printf's to mp_printf's, noting that it bloats code (extra param), and just happy that we can override printf instead, and not touch it. Going in the opposite direction - I'm not sure. Why make it go thru extra layers? All in all, it depends on particular code. For example, in REPL and related funcs, it makes sense vice-versa to use mp_hal_stdout_xxx. At least that's idea I have now. If you have better idea, feel free to go for it.

pfalcon, 2015-11-21:
statvfs is another thing (in uclibc) which pulls fprintf. With socket and statvfs disabled, I'm finally able to get rid of stdio.

pfalcon, 2015-11-21:
But back to issue which was discussed before: floating point printing (conversion to string) should still be handled by libc, because we can't know what format FP uses. Currently, with mips/static uclibc:

~~~
>>> 10000.1
.0
~~~

dpgeorge, 2020-06-09:
Most of the C stdio is removed from the unix port now, so closing this issue.

closed 2020-06-09 by dpgeorge