micropython/micropython #1501: No free space in stmhal port when built without optimizations

pfalcon, 2015-10-11:
@dpgeorge, you from time to time say there's space in stmhal, but actually building it without --gc-sections shows that it overflows, at least in FLASH_ISR section. And any trick like --gc-sections can be buggy/have issues. That's what I see now - just renaming MP_BOOL to mp_obj_new_bool increases text size by 12 bytes. The only explanation would be that rename changes arrangement of all the gazillion of sections, with some useless alignment added, but it's hard to prove that, as building without --gc-sections not possible.

dpgeorge, 2015-10-11:
> just renaming MP_BOOL to mp_obj_new_bool increases text size by 12 bytes

A dirty git state will increase size by 8-12 bytes due to addition of "-dirty" to the build string.  To compare apples with apples I usually add a blank line to a source file, build, use that as the baseline size, then check against size with actual changes.

> it overflows, at least in FLASH_ISR section

The cause and solution of this is "documented" here:

https://github.com/micropython/micropython/blob/master/stmhal/Makefile#L256-L259

pfalcon, 2015-10-11:
D'oh, indeed, with committed changes, it's the same size in both cases.

closed 2015-10-11 by pfalcon