micropython/micropython #1504: RFC: configurablility of object representation, especially with respect to floats

dpgeorge, 2015-10-12:
There is a need to make a more comprehensive selection of object models (ie how uPy objects are represented in a machine word).  They would be configurable at compile time.  In particular there is a need to have more flexibility with floating point representation so floats can be used without the heap.  There are 3 driving use-cases: ESA, microbit, and floats in interrupts on pyboard.

I foresee the following:

1. Floats always stuffed in an object word.

2. Floats stuffed and on the heap.

3. 64-bit NaN boxing.

Float stuffing is similar to int stuffing, except that we need to truncate the mantissa to fit.  It depends on whether the word size is 32-bit or 64-bit, and what precision float you have.  To keep it simple for now, here is the scheme I came up with for single precision float stuffing on a 32-bit machine:
```
iiiiiiii iiiiiiii iiiiiiii iiiiiii1 small int, i's are the 31-bit signed value
01111111 11qqqqqq qqqqqqqq qqqqqq10 str object with 20-bit qstr value
11111111 11qqqqqq qqqqqqqq qqqqqq10 bytes object with 20-bit qstr value (optional)
s1111111 10000000 00000000 00000010 +/- inf (lower 2 bits -> 0)
x1111111 101xxxxx xxxxxxxx xxxxxx10 nan (ok as-is, but could put lower 2 bits -> 0)
seeeeeee efffffff ffffffff ffffff10 30-bit fp, e != 0xff (lower 2 bits -> 0)
pppppppp pppppppp pppppppp pppppp00 ptr (4 byte alignment)
```
The trick is to truncate the last 2 bits of the single precision float (so it's stored as 30-bits) and then use 30-bit nan boxing within that 30-bit float to store qstr strings.

With float stuffing like this, no heap is needed for floating point objects.  Downside is that all floats are stored in 30-bit precision.  This is a loss in precision of about 1 part per million so not a great loss, and would be a good option for some ports that are low on RAM.

Of course, the big advantage of no heap for floats is that you can do floating point operations within an interrupt.

And we can actually keep 32-bit precision outside interrupts by the following trick: we allow 2 different float objects, one is stuffed and the other is the existing heap version.  When extracting a float value from an object you just need to check if it's stuffed or a pointer, it's pretty easy and relatively efficient.  When creating a float object, if the heap is locked then return a stuff float, otherwise allocate on the heap.  So that's option 2 above, stuffed + heap.

The 64-bit NaN boxing option, option 3, is just standard boxing (see eg JavaScript, Lua).  On a 32-bit machine the object size becomes 64-bit.

We could support 64-bit stuffing (ie truncate to 62-bit on a 64-bit machine) and that might be useful for testing etc.

Implementation of stuff floats is relatively easy (I already have a working patch).  64-bit NaN boxing is a bit harder.

One issue will be configuring all this and providing clear config options.  At the moment we just use `MICROPY_OBJ_REPR` with current options being `MICROPY_OBJ_REPR_A` and `MICROPY_OBJ_REPR_B` (latter is for 16-bit systems, but works on 32-bit as well).  Maybe we need more descriptive names?

Comments welcome!

danicampora, 2015-10-12:
I like option 2.

dhylands, 2015-10-12:
+1 for option 2.

dpgeorge, 2015-10-12:
Yes, I agree, option 2 is good for stmhal, and a good default for other ports.  But all options will need to be implemented.

pfalcon, 2015-10-12:
32-bit float truncated on mantissa part by 2 bits means 22-bit mantissa. Well, that still offers better than 10^-6 "base" precision, so supposedly that's ok. But still, what's driving force behind dealing with single-precision floats? Making stmhal port better? And microbit wants to have floating point at all?

Option 2 is more complex than having just FP stuffing, or just FP on heap. Moreover, it introduces inconsistency in how (with what precision) same value can be encoded. This can lead to non-determinism/inconsistent results in programs. And all these issues just to *sometimes* get 2 more bits of precision? Again, it's worth deciding if 22 bits of mantissa is enough for some applications. If so, then just using 22 bits should be enough. Otherwise, using 24 bits (and heap) is the next option.

But again, dealing with single floats is rather partial solution. "Real" floats are double, and NaN-boxing is well-known approach in dealing with them and not compromise on precision. So, for 64-bit floats, it's unclear why "stuffing" option would be selected instead of "NaN-boxing". Except for ease of implementation of course.

Overall, having different options for float representation would be good and exciting, but combining different representations - I'm not sure.

dpgeorge, 2015-10-12:
> 32-bit float truncated on mantissa part by 2 bits means 22-bit mantissa.

Actually, 21 bits (1 sign, 8 exponent, 23 mantissa in single precision).

> But still, what's driving force behind dealing with single-precision floats?

Cortex M4F has single precision hardware floating point, so that leads to single precision being the choice for float operations.  Yes we could do double, but that adds extra code for software add, sub, mul, div, etc, and also makes it slower.

The idea for stmhal is to be able to use FP in an interrupt, and without the use of the heap in general.

> Option 2 is more complex than having just FP stuffing, or just FP on heap. Moreover, it introduces inconsistency in how (with what precision) same value can be encoded. This can lead to non-determinism/inconsistent results in programs.

True.  On the other hand, some people might rely on IEEE standard 32-bit precision float so we don't want to give that up.  But at the same time want to give the option to do FP in an interrupt, and I'd rather be able to do 30-bit in an interrupt than nothing at all.

> Again, it's worth deciding if 22 bits of mantissa is enough for some applications. If so, then just using 22 bits should be enough. Otherwise, using 24 bits (and heap) is the next option.

Option 2 gives this choice to the Python coder: they can use 23-bits exclusively by only doing FP arith when heap is available, or 21-bits by locking the heap (we need a heap-lock function), or a mix if they want.

> But again, dealing with single floats is rather partial solution.

Single precision is all some ports will every need/want/use.

> "Real" floats are double, and NaN-boxing is well-known approach in dealing with them and not compromise on precision. So, for 64-bit floats, it's unclear why "stuffing" option would be selected instead of "NaN-boxing". Except for ease of implementation of course.

Yes, agreed that 64-bit stuffing is not that attractive if proper NaN boxing exists.

On 64-bit archs it makes sense to have 64-bit NaN boxing as the default object model.  On 32-bit archs, I'm not sure.

pfalcon, 2015-10-12:
> Actually, 21 bits (1 sign, 8 exponent, 23 mantissa in single precision).

Well, I follow the idea that one bit is implicit, though I guess right, for precision calculation its should be excluded, and 30-bit float's precision is 1/2 of one millionth of 1.

> Cortex M4F has single precision hardware floating point, so that leads to single precision being the choice for float operations.

Ok, let me say it like that: single-precision floats are toy, and it's worth considering how much effort is spent toying with them ;-).

> The idea for stmhal is to be able to use FP in an interrupt, and without the use of the heap in general.

So, all this float stuffing is partial solution, right? Allowing to use heap during interrupt would allow to use floats as they are implemented now, and more?

> True. On the other hand, some people might rely on IEEE standard 32-bit precision float so we don't want to give that up.

Hopefully not. Just give compile-time option to select which float implementation to use.

> Option 2 gives this choice to the Python coder: they can use 23-bits exclusively by only doing FP arith when heap is available, or 21-bits by locking the heap (we need a heap-lock function), or a mix if they want.

Do people really need such choice at runtime (and if they don't, and don't want to know about it, what surprises it can give them if used)? And beyond that, it yet needs to be implemented, which is more complex than allowing either this or that choice.

> On 64-bit archs it makes sense to have 64-bit NaN boxing as the default object model. On 32-bit archs, I'm not sure.

Only as optional choice, apparently.

Summing up, I think that having 3 clearly separate choices like heap, stuffing, and boxing would be better than 2, one of them combined ;-).

dpgeorge, 2015-10-12:
> Ok, let me say it like that: single-precision floats are toy, and it's worth considering how much effort is spent toying with them ;-).

I do not foresee pyboard having more than single precision, so it apparently is useful in the real world.  Then why not push it to its limits?

> So, all this float stuffing is partial solution, right?

No, it's a final solution.

> Allowing to use heap during interrupt would allow to use floats as they are implemented now, and more?

What's also important is determinism, ie not using the heap at all.  The more you can do without the heap the faster and more deterministic the code.

pfalcon, 2015-10-12:
> Then why not push it to its limits?

The only reason may be that there're lot of interesting/useful things to do, while time is limited. So, if you think it's worth it, sounds good!

> No, it's a final solution.

As in - you decided to never pursue trying to make interrupts being able to use heap?

blmorris, 2015-10-13:
I'm going to agree with both of @pfalcon's points - that 32-bit floats are inadequate for serious number-crunching and that having 30-bit truncated precision floats existing alongside IEEE standard 32-bit precision floats may lead to subtle problems - and still side with wanting this implemented.
If you are doing protein-folding or particle physics calculations (just for example, I'm no expert in either of these), my guess is that the quantization errors introduced by 32-bit floats will quickly lead to unacceptable divergences. Nobody with any sense will be doing these calculations on a bare-metal uPy microcontroller port anyway.

On the other hand, if you are using interrupts to acquire sensor data and want to perform some filtering within the interrupt handler, then even truncated 30-bit floats could provide a significant increase in capability compared to the current integer-only limitation. In digital signal processing, it is a given that the quantization errors introduced by the measurement itself will almost always be larger than the resolution of a 21-bit mantissa (for example, even '24-bit' audio will usually have more than 3 LSB's worth of noise, which will still be completely inaudible.)  The benefit of using floats is that filter coefficients can scale up or down without truncation or significant loss of precision. To accomplish this with integer-only arithmetic usually requires great care and wide data paths

And yes, of course floating-point processing is already available for stmhal, it just needs to be scheduled outside of the interrupt handler. However, another benefit of having truncated floats that don't allocate memory would be to enable time-critical processing that doesn't allocate memory and risk triggering a GC-cycle at an inopportune moment.

For what it's worth, Analog Devices offers the SHARC family of high-end floating-point DSP processors which use 32-bit floats: http://www.analog.com/en/education/education-library/articles/fixed-point-vs-floating-point-dsp.html

peterhinch, 2015-10-14:
@blmorris I agree 100% about DSP.

dpgeorge, 2016-05-06:
Core now has 64-bit NaN boxing and 30-bit float stuffing.

closed 2016-05-06 by dpgeorge

kevinjwalters, 2018-08-27:
Does this 30 bit implementation effectively implement FE_TOWARDZERO rather than FE_TONEAREST?