micropython/micropython #1508: VM issue with MP_BC_WITH_CLEANUP on UNWIND_RETURN

dxxb, 2015-10-13:
The following code
```
class ctx:
    def __enter__(self):
        return self

    def __exit__(self, *exc):
        pass

def test():
    with ctx() as c:
        for l in [1]:
            return

test()
```

Results in `TypeError: 'ctx' object is not callable`

The return statement inside a for loop results in execution reaching https://github.com/micropython/micropython/blob/master/py/vm.c#L576 with an unexpected stack layout (the iterator from the for loop is still on the stack). Which causes the VM to invoke `__call__` on the ctx_mgr object.

pfalcon, 2015-10-13:
Please see if it's the same issue as https://github.com/micropython/micropython/issues/1182

dxxb, 2015-10-13:
Yes, I believe it is. This can be closed.

closed 2015-10-14 by dpgeorge