micropython/micropython #1509: RTC initialisation and wakeup from external events

peterhinch, 2015-10-13:
If the Pyboard is woken from standby by the tamper pin X18, the RTC is reset to 1st January 2014 regardless of any setting performed in code.

It appears impossible to prevent this from Python as it occurs in rtc_init() which is called from main.c. The problem occurs because rtc_init() checks backup register 0 for a specific value 0x32f2 to see if the RTC has already been set. If the value is not correct it assumes the Pyboard has been powered up, sets the RTC to that date, and sets the value of backup register 0 to 0x32f2 to ensure that the RTC continues to run through reset events.

Unfortunately the tamper pin clears the backup registers, so on recovery from standby rtc_init() runs again, detects that the register does not hold 0x32f2 and resets the RTC.

I have produced a build where RTC initialisation occurs in the constructor. This has a boolean argument defaulting True: if present and set False, initialisation is inhibited. This allows the user to take control of the detection of boot events and initialisation of the RTC (power up can be detected by checking the contents of backup RAM which are unaffected by tamper events).

If the argument is absent or True initialisation and boot detection work as at present so existing code should be unaffected.

If this approach meets with approval I can submit a PR.

dpgeorge, 2015-10-13:
So you have rtc_init() called when RTC() is constructed?

Note that the rtc is used also for time.localtime and for fatfs to get the current timestamp when writing a file.  So there would need to be extra code to call rtc_init() from these places as well.

Is there instead a better way to detect that reset has occurred?  I seem to remember there being a few bits in some system reset-control register which tells what the reset source was.

danicampora, 2015-10-13:
Reset causes are part of the new API ;-), and already implemented for WiPy. There's no tamper related one, but feel free to propose it.

peterhinch, 2015-10-14:
@dpgeorge I have now detected RTC power up by reading the date and checking for date.Year == 0. This seems a better solution, avoiding modifying the constructor and merely replacing the test of backup register 0:
```
    HAL_RTC_GetDate(&RTCHandle, &date, FORMAT_BIN);
    // instead of checking data stored in BackUp register0
    if (date.Year == 0) {
        // fresh reset; configure RTC Calendar
        RTC_CalendarConfig();
    } else {
```
It allows the RTC to run through soft and hard resets.

peterhinch, 2015-10-14:
@danicampora The Tamper and WKUP pins resume from standby rather than causing a reset: should they be included in the Reset Cause API? Tamper and timed wakeup can be identified as reasons for resumption by reading the RTC_ISR. The only way I've so far found to identify WKUP is by the absence of other causes. There must be a better way...

danicampora, 2015-10-14:
If they don't reset, then should be classified as "wake reasons" instead of "reset causes". The former is also part of the new API.

danicampora, 2015-10-14:
@peterhinch after reading your first post again, it looks like waking up from standby starts execution from the reset vector, is that right? in that case I would say it is indeed a `reset cause` and would correspond to `machine.DEEPSLEEP_RESET`  in the new API.

peterhinch, 2015-10-14:
@danicampora I don't know the answer to that one I'm afraid.
@dpgeorge There is the RCC_CSR which provides information about the source of a reset. I still think the most logical way to check for an unconfigured RTC is to check for a year value of zero and this approach does seem reliable.