micropython/micropython #1510: Unix build fails on ARM

ha5di, 2015-10-14:
Building the currint git clone for Unix on Raspberry Pi2 fails, see below. Last build on September 24 was OK.

CC ../py/emitnative.c
../py/emitnative.c: In function 'emit_native_load_subscr':
../py/emitnative.c:489:52: error: too few arguments to function 'asm_arm_ldr_reg_reg'
 #define ASM_LOAD32_REG_REG(as, reg_dest, reg_base) asm_arm_ldr_reg_reg((as), (reg_dest), (reg_base))
                                                    ^
../py/emitnative.c:1524:21: note: in expansion of macro 'ASM_LOAD32_REG_REG'
                     ASM_LOAD32_REG_REG(emit->as, REG_RET, reg_base); // load from (base+4*index)
                     ^
In file included from ../py/emitnative.c:407:0:
../py/asmarm.h:113:6: note: declared here
 void asm_arm_ldr_reg_reg(asm_arm_t *as, uint rd, uint rn, uint byte_offset);
      ^
../py/emitnative.c:489:52: error: too few arguments to function 'asm_arm_ldr_reg_reg'
 #define ASM_LOAD32_REG_REG(as, reg_dest, reg_base) asm_arm_ldr_reg_reg((as), (reg_dest), (reg_base))

dpgeorge, 2015-10-14:
Should now be fixed.  Please close this issue if it is.

ha5di, 2015-10-14:
Builds fine now, issue fixed.

closed 2015-10-14 by ha5di