micropython/micropython #1515: Implement struct.pack_into?

dpgeorge, 2015-10-17:
I saw on the forum (http://forum.micropython.org/viewtopic.php?f=6&t=953&p=5816#p5816) a desire for struct.pack_into and remembered that there are a few such "_into" methods that we thought we be good to implement.  struct.pack_into seems like it would be well used, and it's a relatively small addition (can use existing pack code).

Is it something we want to add?

I would suggest placing the buffer before the type-spec, so that the buffer doesn't get confused with one of the arguments to pack, ie:
```python
struct.pack_into(buf, 'iii', 1, 2, 3)
```
Should it be called `pack_into` or `packinto`?

pfalcon, 2015-10-17:
There's already official struct.pack_into() method: https://docs.python.org/3/library/struct.html#struct.pack_into

But I don't think it's worth implementing it. For example, what forum post you quote does is wrapping uctypes.struct into custom class (and struct too, by extension). But this wrapping is not required, uctypes.struct can be used directly to get syntax like `bkram.setting1`, `bkram.setting2`.

So, I'd really wish people use uctypes more instead. And what we need to do to make it easier to use, is finally implement automatic field offset calculation now that OrderedDict is implemented.

dpgeorge, 2015-10-17:
> There's already official struct.pack_into() method

True!

> But this wrapping is not required, uctypes.struct can be used directly to get syntax like bkram.setting1, bkram.setting2.

I see your point, and, yes, proper use of uctypes is ideal for that use case.

> So, I'd really wish people use uctypes more instead. And what we need to do to make it easier to use, is finally implement automatic field offset calculation now that OrderedDict is implemented.

Also use in some example code would teach people of its existence.  I think the main problem is that people don't know about it.

Regardless, I still think struct.pack_into is useful.  But I'm happy to wait for a few more "requests" before implementing it.

pfalcon, 2015-10-18:
> Also use in some example code would teach people of its existence. I think the main problem is that people don't know about it.

Well the issue is that uctypes isn't ideally suited just everything, where it shines is: 1) automatic conversion of C headers; 2) dealing with many bitfields. As you may imagine, both are things someone deals with everyday (in regard to uPy). I made some movement in 1st direction with https://github.com/pfalcon/micropython-ffigen , but not much with actual usage of its results. And for 2nd use, I thought @dhylands would have progress, but I guess presence of "stm" module effectively kills any motivation.

> Regardless, I still think struct.pack_into is useful. But I'm happy to wait for a few more "requests" before implementing it.

It would be nice to stick to the known "Django criteria" - if that's the last thing needed to run some known and useful 3rd-party Python code, then yes, makes sense. For code written from scratch, it makes sense to just use other functionality.

dbc, 2015-10-21:
So the comment:
```python
# Wouldn't it be nice to have struct.pack_into(...)?  Oh, well. 
```
is in some example code I posted for accessing the backup SRAM.  I have to agree with @pfalcon here, the case for making ROM space for pack_into() is pretty thin.  Sure, once in a while it could save a few bytes of byte code, but on balance I think the ROM is better spent elsewhere.

(And yes, ucstruct could have been used directly in the example code.  I have an example for that, too, but thought the thread should be left to die a natural death...)

dpgeorge, 2015-10-21:
@dbc thanks for the feedback!

> Sure, once in a while it could save a few bytes of byte code, but on balance I think the ROM is better spent elsewhere.

Well, I'd counter with the following: ROM is almost unlimited on pyboard so we can add such features without a problem.  And what it buys you is not making the code shorter, but rather the ability to use struct.pack_into in an interrupt, where you can't allocate memory.

dbc, 2015-10-22:
@dpgeorge 
> And what it buys you is not making the code shorter, but rather the ability to use struct.pack_into in an interrupt, where you can't allocate memory.

OK, so that is compelling.  One of the first things that I bumped into with the PyBoard is how to get a counter value out of a 32 bit quadrature encoder.  That spawned issue #1454.  I think the outcome of that issue is that 32 bit peripherals can turn long ints into 32 bit machine ints without too much trouble, but a callback would have trouble reading a 32 bit counter atomically because it can't allocate memory, and stopping interrupts doesn't stop counters, so somehow you need to turn a 32 bit atomic register read into something that Python code can see.  In the mean time, I'm using a 31 bit counter because the batteries in my robot will be flat before the counter rolls over, but in the general case the issue of doing an atomic read of a 32 bit peripheral register is still out there.

Although... doesn't pack_into go the other way?  For a register read, I guess you would combine a ucstruct on the register as input to a pack_into().

peterhinch, 2015-10-22:
@dbc Does section 1.6 [here](http://docs.micropython.org/en/latest/reference/asm_thumb2_hints_tips.html) offer an (admittedly arguably hacky) solution to your immediate problem?

dbc, 2015-10-22:
@peterhinch The solution you propose could be made to work as long as the work can be split across interrupt safe and non-interrupt-safe sections and managed appropriately. My previous post was made in haste to get out the door and reflects some fuzzy thinking, in that pack_into isn't going to be interrupt safe if gathering the input requires an allocate.  I think I am correct in thinking that reading a 32 bit counter in Python to pass it to pack_into would do an allocation.  Likewise, inserting a ucstruct access as a parameter to pack_into still generates an intermediate Python long int.  In some ways it is as if we need a version of ucstruct that works like a Linux driver with "top half" and a "bottom half".  The top half would do an interrupt-safe capture into a pre-allocated buffer, and the bottom half could read from that and create Python allocations.  Of course that brings with it the whole world of critical sections and buffer overwrites, not to mention having to poll to see if the bottom half should run.

pfalcon, 2015-10-31:
With all the talks to add more stuff t add to ustruct, I actually went and implemented https://github.com/micropython/micropython/commit/2b080cf599c3734a66b719fc6d0b59ea9fe15caf , that's more useful change imho, to avoid need to patch cpython code.

dpgeorge, 2016-01-29:
struct.pack_into and struct.unpack_from were implemented in a17755ee8ba2982505d349e0f06acdabe943d33e.

closed 2016-01-29 by dpgeorge