micropython/micropython #1518: Prepare for 1.4.7/1.5 relase

pfalcon, 2015-10-18:
We've got few great features/fixes implemented lately, and overall 100+ commits and ~ a month past the last release, so it's probably time to think about new release. And I really think that recent features/fixes were great, to call it 1.5 release. However, unlike previous times, not all features as fully finished/merged yet, and more great features are in the queue, so the more important it's to make good release planning, to see what fits, and to not block merging of next features.

dpgeorge, 2015-10-19:
Yes, we can go to 1.5.  Main features would be:
- modussl
- standalone unix binary
- machine module for WiPy
- would be good to merge start of machine module for stmhal: #1493
- would be nice to merge stuff float support: #1516

Anything else?

pfalcon, 2015-10-19:
* Top-notch REPL support (even if actual change is paste mode addition, it's now complete and very usable)
* Can mention JNI integration module
* I wanted to make https://github.com/micropython/micropython/pull/1485 in, now merged, but maybe @danicampora will want to refactor wipy impl.

#1516 might wait for after release I'd think - just to not put too much goodies in changelog ;-). If you think it's important for it to be in, well...

So, regarding modussl, I'd like to add "make deplibs" to Travis - at least, better to also build upy with MICROPY_PY_USSL=1, but that will smash coverage numbers ;-). Still not enabled by default when cloned. But README has info how to build it.

fojie, 2015-10-19:
@dpgeorge does the banner line still say "Micro Python"? If so would this be a good time to update it to "MicoPython" according to decision(?) half a year ago?

danicampora, 2015-10-19:
> I wanted to make #1485 in, now merged, but maybe @danicampora will want to refactor wipy impl. 

Done in: https://github.com/micropython/micropython/commit/ae70e98ed4cae8bc6c919abd0393e16980616889

pfalcon, 2015-10-19:
@danicampora : Nice, thanks!

> I'd like to add "make deplibs" to Travis

In https://github.com/micropython/micropython/pull/1521

dpgeorge, 2015-10-19:
> better to also build upy with MICROPY_PY_USSL=1

Yes, we can do this.  I'd suggest making micropython_coverage built with ussl, ie change the Makefile to make this happen.

> but that will smash coverage numbers

We can deal with that later :)

dpgeorge, 2015-10-19:
> does the banner line still say "Micro Python"

Nope, it now says "MicroPython" :)

Regarding REPL, and the second line of the banner that now says "Use CTRL-D to exit, CTRL-E for paste mode".  I kind of don't like having CTRL in caps, it "yells" a bit too much for my liking, especially running uPy on the command line over and over.  I'd prefer to change it to all lower case, ie "Use ctrl-D to exit, ctrl-E for paste mode".  Am I being too picky?

pfalcon, 2015-10-19:
> ie "Use ctrl-D to exit, ctrl-E for paste mode". Am I being too picky?

I don't know if I'm too picky lately about all that lower case. Look at your keyboard - it's "Ctrl" ;-)

dpgeorge, 2015-10-19:
> Look at your keyboard - it's "Ctrl" ;-)

On my keyboard it's not there because I have one of these https://en.wikipedia.org/wiki/Happy_Hacking_Keyboard with blank key tops :)

pfalcon, 2015-10-19:
Ok, then let's approach it from that side that in English, names (and even titles) starts with capital letter, and that's name of keys, so it should be Ctrl-D (and arguably, it should be "+" to emphasize pressing at the same time)

dpgeorge, 2015-10-19:
> and that's name of keys

Arguably it's not a key but rather a character, or command, being chr(4).  And its name is "Control-D" (or EOF).  It just so happens that you press the "Ctrl" key and the "D" key together to generate it.

https://en.wikipedia.org/wiki/Control-C

pfalcon, 2015-10-19:
The message as printed there, isn't technical, but rather a mundane notice of which keyboard keys should be pressed together for this or that action.

dpgeorge, 2015-10-19:
My point was that they spell it "ctrl-D" etc (ie without plus).

pramasoul, 2015-10-19:
^S

dhylands, 2015-10-19:
Don't forget to fix pyboard.py to match the final decision.

dpgeorge, 2015-10-19:
> Don't forget to fix pyboard.py to match the final decision.

No, I don't think we'll change the raw/friendly REPL printing in pyexec.c.  Just the unix one.  Otherwise it'll lead to difficulties with different versions of pyboard.py vs stmhal firmware.

danicampora, 2015-10-20:
I have two HP keyboards in front of me and both read `ctrl` ...

dhylands, 2015-10-20:
The 2 Mac keyboards (one for iPad, one for MacBook) both say 'control'

The 2 HP laptops I have say 'ctrl'

My Lenovo ThinkPad, System 76 laptop and all of my Dell laptops all say 'Ctrl'.

My Microsoft keyboard and all of my Dell keyboards say 'Ctrl'.

My Logitech keyboard says 'CTRL'

pfalcon, 2015-10-20:
> My point was that they spell it "ctrl-D" etc (ie without plus).

People who think about *ware being accessible for 11-years old use plus ;-) : 

https://youtu.be/HAx9sJB1_eY?t=150

pfalcon, 2015-10-20:
In the meantime, 8e6e9eaea575f2f9b7b3591c813002cd94ff45e3

pfalcon, 2015-10-20:
So, https://github.com/micropython/micropython/pull/1522 is the last thing I have in my queue for release.

dpgeorge, 2015-10-20:
I just got esp8266 building again, and I'd also like to get all tests either properly skipping or passing on pyboard.

dpgeorge, 2015-10-20:
All tests now work (or skip) on pyboard.  So I think I'm done.

pfalcon, 2015-10-21:
Nice, let's tag tomorrow and start merging new exciting stuff? ;-)

@dpgeorge , if you'll send announcement, I'd suggest spamming python-users and python-dev this time in addition to *-announce ;-). Or otherwise I'd plan to do that (when time permits).

danicampora, 2015-10-21:
I have 2 tests failing on WiPy, I couldn't take a look at it yesterday, but will do today.

dpgeorge, 2015-10-21:
I will prepare release 1.5.

@danicampora please let me know when the tests succeed.

danicampora, 2015-10-21:
I will, thanks!

stinos, 2015-10-21:
Note that the msvc port currently doesn't compile due to lack of ```usleep```. Not too hard to fix (though actual uSec resolution is not available afaik) but I don't think I will have time until next monday. If that is too long of a wait for you guys go ahead and release anyway, it's not like thousands are standing in line waiting for this port :P

dpgeorge, 2015-10-21:
Perhaps we should disable time.sleep_ms and time.sleep_us for _WIN32, just for now to get it building?

pfalcon, 2015-10-21:
> though actual uSec resolution is not available afaik

How is that possible?

> If that is too long of a wait for you guys go ahead and release anyway, it's not like thousands are standing in line waiting for this port :P 

Indeed, we don't (yet) have those formal release schedule, so releases as they are give various chance to polish their stuff, not require that. We'll have another release in a month (and for it, more issue than you know now will need to be tackled anyway).

pfalcon, 2015-10-21:
> I just got esp8266 building again

@dpgeorge : Btw, did you use latest esp-open-sdk (with vendor SDK 1.4.0)? I still didn't have that tested beyond compiling...

dpgeorge, 2015-10-21:
> did you use latest esp-open-sdk

No, I'm using old 1.1.2.

pfalcon, 2015-10-21:
Any reason for that?

dpgeorge, 2015-10-21:
> Any reason for that?

Simply lack of time.

danicampora, 2015-10-21:
I fixed my tests and pushed, go ahead when you are ready.

dpgeorge, 2015-10-21:
Ok, no pushing until 1.5!

dpgeorge, 2015-10-21:
Done.

closed 2015-10-21 by dpgeorge

pfalcon, 2015-10-21:
@dpgeorge: Thanks, and congrats! ;-)

dpgeorge, 2015-10-21:
@pfalcon I think congrats to us all, it's a team effort :)  Even those who don't contribute direct code still contribute to the discussions, and submit great but reports, etc.

BTW, JNI was announced in 1.4.6 release, but we can still advertise it in announcements.

pfalcon, 2015-10-21:
I don't think 1.4.6 was anyhow widely announced, so if looking for "highlights", it can be mentioned.