micropython/micropython #1519: cc3200: socket.makefile() needs to be more CPython compliant

dpgeorge, 2015-10-19:
socket.makefile() has the following spec:
```python
socket.makefile(mode='r', buffering=None, *, encoding=None, errors=None, newline=None)
```
The main thing is that it should return a file in text mode by default.  But cc3200's socket.makefile takes no args and just returns the socket which is in bytes mode.

unix port has the correct behaviour, by delegating the makefile to the file "open" function.  But that only works because of file descriptors.

At the very least I think cc3200 should accept the mode argument and check that it is 'rb' or 'wb' and raise an exception if it's not (meaning that text mode is not supported).  Alternatively, a full implementation would require a new socket type that was in text mode, which adds a lot to the code size (maybe there is a better way?).

danicampora, 2015-10-19:
> At the very least I think cc3200 should accept the mode argument and check that it is 'rb' or 'wb' and raise an exception if it's not (meaning that text mode is not supported).

I would go for this option for now.

dpgeorge, 2017-06-14:
In the end esp8266 (actually extmod/modlwip) just implements a dummy makefile() method which returns the socket object.  So in 696fcde8009b3670e5c4e867600c17a916f9a3b0 I made cc3200 have the same behaviour, which saves a few precious bytes of code.

closed 2017-06-14 by dpgeorge