micropython/micropython #1523: interaction between pyb.standby() and pyb.rtc()

jimtal001, 2015-10-20:
In my attempt to calibrate the RTC using rtc.calibration (), I found that there is a bug related to pyb.standby().  To reproduce the issue use the following code (time_cal.py) to set, syncronize and output the time to PC for a one hour period.  At the end of the one hour period I found approximately a 19,000 ppm (tick) delta between PC and pyboard.  The mode pyb.stop() does NOT have this issue and adding a battery backup VBAT does not fix the issue for pyb.standby().

```
import pyb, stm
import micropython

rtc = pyb.RTC()
sleepTime = 10000 # in ms

def main():
   if (stm.mem32[stm.RTC + stm.RTC_BKP1R] == 0): # first boot, Code to run on 1st boot only
      rtc.datetime((2015,10,19,1,16,45,0,0))
      print(rtc.datetime())
      
   print(rtc.datetime())
   rtc.wakeup(sleepTime)
   stm.mem32[stm.RTC + stm.RTC_BKP1R] = 1 # indicate that we are going into standby mode
   pyb.standby()
   
main()
```
python

This is my boot.py
```
import pyb
import micropython
pyb.usb_mode(None)
uart = pyb.UART(4,115200)
pyb.repl_uart(uart)
pyb.main('time_cal.py')
```
python

dpgeorge, 2015-10-20:
This is strange because the RTC should run independently of any software that controls it.  It's a separate peripheral with its own crystal and should just tick away, regardless of stop or standby mode.

jimtal001, 2015-10-20:
Using stop() mode the RTC time remains reliable.  Using standby() mode (perhaps with every wakeup event) there is a loss of time.  I measured around 0.18 sec per wake event.  This issue seriously impacts data logger type applications where time of day is an important parameter.

dhylands, 2015-10-21:
@dpgeorge my guess is that something during initialization is causing the
time loss (so when coming out of reset)

peterhinch, 2015-10-21:
I'm running a test to try to reproduce this and will report back. A test to check @dhylands hypothesis will be to power cycle a Pyboard (with RTC battery) every ten seconds for an hour. I'll rig something up to try that once this check is complete.

chuckbook, 2015-10-21:
With current firmware, wakeup from standby requires 32KHz osc. initialization. This takes around 150 ms.

dpgeorge, 2015-10-21:
@dhylands @chuckbook yes, you're right: exiting from standby initiates a "power reset" which resets all registers except those in the "backup domain".  The backup domain includes the RTC and the backup SRAM.

So, we need to detect a power reset and in this case not call rtc_init.

danicampora, 2015-10-21:
That what I suggested before to @peterhinch

peterhinch, 2015-10-21:
@danicampora Sorry, it seems I failed to pick up on the significance of your comment about interrupt vectors. For what it's worth I've confirmed the observations of @jimtal001.

chuckbook, 2015-10-21:
@dpgeorge as the stm32f4xx RTC oscillator is well known to be tricky (to be polite), we ended up with following strategies:
1. low cost --> get rid of LSE use LSI instead (good enough for casual use)
2. regular --> replace crystal with 32KHz TCXO (low power versions are available)
3. 24/7 usage --> use HSE or even better, ext. TCXO
We successfully modified the MPY rtc subsystem to allow rtc_init to check availability of a proper LSE or LSI clock before trying RTC initialization. Handling the variants mentioned above might be an overkill but it works for us.

dpgeorge, 2015-10-21:
@chuckbook would you consider posting your modifications to rtc_init for inclusion into the main code?

chuckbook, 2015-10-21:
@dpgeorge no problem, give me some time to strip down the code for the pyboard. I guess the HSE mode is not needed as it would require additional mods outside rtc_init.

dpgeorge, 2015-10-21:
@chuckbook just the LSE/LSI detection code would be great, but anything else also much appreciated!

dhylands, 2015-10-21:
> I'm running a test to try to reproduce this and will report back. A test to check @dhylands hypothesis will be to power cycle a Pyboard (with RTC battery) every ten seconds for an hour. I'll rig something up to try that once this check is complete.

Or just do a pyb.hard_reset()

chuckbook, 2015-10-21:
@dpgeorge here the stripped down patch (with LSI fallback)

dpgeorge, 2015-10-21:
@chuckbook I don't see any link/url...?

chuckbook, 2015-10-21:
--- ../../micropython/stmhal/rtc.c	2015-10-21 12:15:08.769162363 +0200
+++ ./rtc.c	2015-10-21 18:26:07.188137077 +0200
@@ -47,13 +47,19 @@
 
 // rtc_info indicates various things about RTC startup
 // it's a bit of a hack at the moment
-static mp_uint_t rtc_info;
+static mp_uint_t rtc_info = 0x0;
 
 // Note: LSI is around (32KHz), these dividers should work either way
 // ck_spre(1Hz) = RTCCLK(LSE) /(uwAsynchPrediv + 1)*(uwSynchPrediv + 1)
 #define RTC_ASYNCH_PREDIV (0x7f)
 #define RTC_SYNCH_PREDIV  (0x00ff)
 
+#if defined(MICROPY_HW_RTC_USE_LSE) && MICROPY_HW_RTC_USE_LSE
+uint32_t rtc_use_LSE = 1;
+#else
+uint32_t rtc_use_LSE = 0;
+#endif
+
 #if 0
 #define RTC_INFO_USE_EXISTING (0)
 #define RTC_INFO_USE_LSE (1)
@@ -179,15 +185,54 @@
     RTCHandle.Init.OutPutType = RTC_OUTPUT_TYPE_OPENDRAIN;
 
     mp_uint_t tick = HAL_GetTick();
+    /* wait 5 ms (RTC_SSR should change) */
+    uint32_t ss = RTC->SSR;
+    while (HAL_GetTick() != (tick+5)) {
+        __WFI();
+    }
+    if (ss != RTC->SSR) {
+        // we have a running RTC...
+        // should be sufficient for now...
+#ifdef RTC_HSE_MIRROR
+        need_sync = 1;
+#endif
+        // Clear source Reset Flag
+        __HAL_RCC_CLEAR_RESET_FLAGS();
+        rtc_info |= 0x40000;
+        return;
+#ifdef FORCE_RTC_UNTOUCHED
+    } else {
+        need_sync = 1;
+        // Clear source Reset Flag
+        __HAL_RCC_CLEAR_RESET_FLAGS();
+        rtc_info |= 0x80000;
+        return;
+#endif
+    }
 
+    uint32_t rtc_ok = 0;
     if (HAL_RTC_Init(&RTCHandle) != HAL_OK) {
-        // init error
-        rtc_info = 0xffff; // indicate error
+        if (rtc_use_LSE != 0) {
+            // fall back to LSI...
+            rtc_use_LSE = 0;
+            if (HAL_RTC_Init(&RTCHandle) == HAL_OK) {
+                rtc_info |= 0x100000;
+                rtc_ok = 1;
+            } else {
+                // not even LSI...
+            }
+        }
+    } else {
+        rtc_ok = 1;
+    }
+    if (rtc_ok == 0) {
+         // init error
+        rtc_info |= 0xffff; // indicate error
         return;
     }
 
     // record how long it took for the RTC to start up
-    rtc_info = HAL_GetTick() - tick;
+    rtc_info |= (HAL_GetTick() - tick) & 0xffff;
 
     HAL_RTC_GetDate(&RTCHandle, &date, FORMAT_BIN);
     if (date.Year == 0 && date.Month ==0 && date.Date == 0) {
@@ -259,24 +304,25 @@
 
     RCC_OscInitStruct.OscillatorType =  RCC_OSCILLATORTYPE_LSI | RCC_OSCILLATORTYPE_LSE;
     RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
-    #if defined(MICROPY_HW_RTC_USE_LSE) && MICROPY_HW_RTC_USE_LSE
-    RCC_OscInitStruct.LSEState = RCC_LSE_ON;
-    RCC_OscInitStruct.LSIState = RCC_LSI_OFF;
-    #else
-    RCC_OscInitStruct.LSEState = RCC_LSE_OFF;
-    RCC_OscInitStruct.LSIState = RCC_LSI_ON;
-    #endif
+    if (rtc_use_LSE) {
+        RCC_OscInitStruct.LSEState = RCC_LSE_ON;
+        RCC_OscInitStruct.LSIState = RCC_LSI_OFF;
+    } else {
+        RCC_OscInitStruct.LSEState = RCC_LSE_OFF;
+        RCC_OscInitStruct.LSIState = RCC_LSI_ON;
+    }
+
     if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
         //Error_Handler();
         return;
     }
 
     PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_RTC;
-    #if defined(MICROPY_HW_RTC_USE_LSE) && MICROPY_HW_RTC_USE_LSE
-    PeriphClkInitStruct.RTCClockSelection = RCC_RTCCLKSOURCE_LSE;
-    #else
-    PeriphClkInitStruct.RTCClockSelection = RCC_RTCCLKSOURCE_LSI;
-    #endif
+    if (rtc_use_LSE) {
+        PeriphClkInitStruct.RTCClockSelection = RCC_RTCCLKSOURCE_LSE;
+    } else {
+        PeriphClkInitStruct.RTCClockSelection = RCC_RTCCLKSOURCE_LSI;
+    }
     if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK) {
         //Error_Handler();
         return;

peterhinch, 2015-10-22:
Perhaps the following observation is significant. While qualitatively my testing confirmed the observations of @jimtal001, there were substantial quantitative differences with standby. Stop worked perfectly within the limits of my measurement - +-1 sec per hour.

I tested by modifying existing code which, on recovery from standby, loads a substantial amount of Python code before reading the RTC and displaying the result on an ePaper screen. There is some 8 seconds of activity while it boots and runs code before it sets a 10 second delay and enters standby. Over an hour's running I experienced a time loss of approximately 1.25 seconds per update.

This seems to imply that the RTC is stopped for some time after recovery from standby and that the time is application dependent. Time spent loading and compiling code, perhaps?

Diagnosing this is frankly beyond me but I'm happy to do any testing on request.

manitou48, 2015-10-23:
"So, we need to detect a power reset and in this case not call rtc_init."

rtc.c  does if (HAL_RTCEx_BKUPRead(&RTCHandle, RTC_BKP_DR0) != 0x32f2) 
to determine if RTC was previously set, would that work?

chuckbook, 2015-10-23:
I made a proposal which solves the lost RTC clock cycles problem. (see my previos patch)
The LSI fallback is probably not needed for the pyboard.

peterhinch, 2015-10-24:
@chuckbook With ``pybv10-2015-10-21-v1.4.6-143-g9d7ef05.dfu`` the following coe behaves as expected, with the green LED briefly flashing every ten seconds and power consumption below 1mA while in standby.
```python
import pyb
rtc = pyb.RTC()
rtc.wakeup(10000)
pyb.standby()
```
With the modified rtc.c, after power-up current consumption is initially low, implying that the first standby() has worked. After 10 seconds the green LED flashes once. Thereafter the Pyboard consumes 40mA and no further activity occurs. At the REPL the RTC responds as expected, but standby() appears not to be working consistently.

chuckbook, 2015-10-24:
@peterhinch thanks for the hint, stripped down too much...
after a non-powerup reset the backup domain is protected. A
PWR->CR |= PWR_CR_DBP;
would fix that. This could be done in rtc.c at:
 // Clear source Reset Flag
PWR->CR |= PWR_CR_DBP;   // <<<<<  Disable backup domain write protection
__HAL_RCC_CLEAR_RESET_FLAGS(); 
rtc_info |= 0x40000;

peterhinch, 2015-10-25:
@chuckbook I'm still seeing the same symptom I'm afraid.

chuckbook, 2015-10-25:
@peterhinch It works for me. Where did you place the code to unlock backup domain?

chuckbook, 2015-10-26:
@dpgeorge  @peterhinch 
I spent some time to trim our rtc.c patch for the pyboard. LSI fallback is not supported anymore.
Valid RTC is defined an enabled & ready LTE oscillator.
Minimum standby / wakeup cycle is less than 100ms (at 168MHz freq) using the test from @peterhinch . Average current consumption is around 340µA (10s interval).
Of course no loss in RTC time.

I would love to see this modification in micropython, even if it does not fulfill all of our requirements.    

--- ../micropython/stmhal/rtc.c 2015-10-21 12:15:08.769162363 +0200
+++ stmhal/rtc.c        2015-10-26 10:22:16.606416952 +0100
@@ -180,6 +180,17 @@
 
     mp_uint_t tick = HAL_GetTick();
 
+    // if LTE enabled & ready --> no need to (re-)init RTC
+    if ((RCC->BDCR & 3) == 3) {
+        // remove Backup Domain write protection
+        PWR->CR |= PWR_CR_DBP;
+        // Clear source Reset Flag
+        __HAL_RCC_CLEAR_RESET_FLAGS();
+        // provide some status information
+        rtc_info |= 0x400000 | (RCC->BDCR & 7) | (RCC->CSR & 3) << 8;
+        return;
+    }
+

peterhinch, 2015-10-26:
This seems to do the trick. In an hour's running, waking every ten seconds, the RTC kept time within the limits of my measurement (+- 1sec). It also survives wakeups from the Tamper and WKUP pins. I haven't proved whether time is kept under those circumstances: I suggest we get the code reviewed before doing any more testing.

@chuckbook It would be good to have this fixed. The way forward is for you to fork the project and submit a PR so that the code can be reviewed. If you're new to Github I suggest reading [this](https://github.com/micropython/micropython/wiki/DevelWorkflow) as a guide to how to proceed.

dpgeorge, 2015-10-26:
@peterhinch so the only change you made to latest master branch is add the 11 lines in the patch directly above?

peterhinch, 2015-10-26:
No, I assumed all the prior changes were also required. Perhaps @chuckbook can clarify?

chuckbook, 2015-10-26:
@dpgeorge  & @peterhinch 
It is just the 11 lines patch, which I made against
MicroPython v1.5-16-gb7ab70c-dirty on 2015-10-25; PYBv1.0 with STM32F405RG
this morning.
As mentioned before, we are investigating using mpy in real applications (mainly STM32F7xx), where timekeeping, fault handling etc. is critical.
It will take some time to prepare a branch for plain pyboard where we can contribute using the recommended pocedure.
Following the RTC discussion for quite some time now I finally decided to present our solution for this tiny problem, knowing it is not the recommended procedure for code contribution.
There are some more patches waiting for "tiny" quirks in micropython (date handling before 1-mar-2000, ADC calibration, Temperature handling etc.)

dpgeorge, 2015-10-26:
@chuckbook I'd much rather have a patch submitted simply as a diff (like you did above) than have no patch at all, especially for these corner-cases and quirks, which can take a long time to find and fix.  So, please open an issue and paste your diff if that's the only realistic way for you.  Then we can test it and maybe someone else can open a PR based on your diff.  Thanks!

peterhinch, 2015-10-26:
@dpgeorge I'm happy to open a PR if this is how @chuckbook wants to proceed.

@chuckbook please let us know if you want to go that way. If so also confirm that the diffs you've provided to date need to be applied cumulatively. Alternatively send a copy of rtc.c to
peter at hinch dot me dot uk

dpgeorge, 2015-10-30:
This should now be fixed by 38196344691ad32b7f80bb44489c069ceaecb1ec.

@jimtal001 are you able to verify that it now works for you?

jimtal001, 2015-11-04:
dpgeorge, 
Sorry for the delay.  Is the fix in the latest dfu?

dpgeorge, 2015-11-04:
@jimtal001 yes the fix should be in the latest dfu.

jimtal001, 2015-11-05:
After a 17 hour test I can now say the issue is closed.  I got a 129 ppm offset.  Thanks for all the great work.

closed 2015-11-05 by jimtal001