micropython/micropython #1524: cc3200: FTP server crashes when secure connection attempted

dpgeorge, 2015-10-22:
The FTP server in cc3200 seems to crash when TLS authentication is attempted by the client.  See forum for some details: http://forum.micropython.org/viewtopic.php?f=11&t=1052&p=6140#p6137 ; in particular the commands between the client and server are:
```
<--- 220 Micropython FTP Server
---> FEAT
<--- 332                         
---> AUTH TLS
<--- 332                       
---> USER micro
```
At this point the server stops responding.

Probably not the cause, but there is a bug at https://github.com/micropython/micropython/blob/master/cc3200/ftp/ftp.c#L615 : the size of the ftp_cmd_buffer is FTP_MAX_PARAM_SIZE + FTP_CMD_SIZE_MAX, so FTP_BUFFER_SIZE in this line should be that expression (which is about 130 bytes, vs 512 for FTP_BUFFER_SIZE).

Also, is it guaranteed that an incoming command from the client will always be terminated by a newline?  Because the command scanning code assumes this.

danicampora, 2015-10-22:
Thanks Damien, I'll investigate.

dpgeorge, 2015-10-22:
The responses from the server are 332 for the FEAT and AUTH commands because there is no valid user logged in at that point.  The first command that actually gets processed by the server is the USER command, and that's the one that crashes.

danicampora, 2015-10-27:
@dpgeorge https://github.com/micropython/micropython/commit/a0fb7a76cd335ad4cae118a0fec731d748ec94c6 fixes the bug you spotted and also the crash when attempting more than 1 data connection. I can't reproduce the TLS issue with FileZilla (the connection fails but the WiPy doesn't crash), and with lftp I haven't found the way to enable TLS. Are you able to reproduce this?

dpgeorge, 2015-10-27:
I tried lftp and indeed it crashed the WiPy when trying to log in.  BUT, it seems to be fixed with the patch mentioned above.  Specifically, I can now use lftp as follows:
```
$ lftp -u micro,python 192.168.0.24
lftp micro@192.168.0.24:~> set ftp:passive-mode 
lftp micro@192.168.0.24:~> ls
drw-rw-r--   1 root  root         0 Jan  1 00:00 flash
lftp micro@192.168.0.24:~>
```
I think it automatically tries TLS first, and now seems to work.  It was either a buffer overflow, or trying to connect multiple times.

danicampora, 2015-10-28:
> It was either a buffer overflow, or trying to connect multiple times.

OK that would explain it. I'll post a binary with the patch in the forum so that people can try it and give feedback. Thanks!

dpgeorge, 2015-10-28:
> I'll post a binary with the patch in the forum so that people can try it
and give feedback.

So we don't overload the forum will lots of large attachments, is it
possible to put the firmware at wipy.io and link to it? Or at
micropython.org/download with the other firmware? I can easily add cc3200
to the list of firmwares that are built automatically each day (at 00:10
GMT).

danicampora, 2015-10-28:
> Or at
micropython.org/download with the other firmware? I can easily add cc3200
to the list of firmwares that are built automatically each day (at 00:10
GMT).

That would be perfect!

dpgeorge, 2015-10-28:
Ok, wipy is added to auto-build procedure and now available at micropython.org/download

The firmware and docs are now built twice a day, at 00:10 and 12:10.  When I get time I'll make it update on git push.

danicampora, 2015-10-28:
Thanks Damien!

dpgeorge, 2018-06-20:
It seems this was fixed long ago.

closed 2018-06-20 by dpgeorge