micropython/micropython #1525: RFC Implementing platform specifics using existing C++ code

stinos, 2015-10-23:
Just an idea: the msvc C runtime library isn't POSIX so a bunch of functions need a custom implementation. However some of this functionality is available in the C++ runtime library so question is in these cases should we reinvent the wheel, or just use the available implementation? 

Currently take usleep() for instance: I can either go search and copy-paste + test an implementation found on the net or in source code somewhere (or be lazy like mingw and implement it as ```Sleep( msec / 1000 )```), or use ```std::this_thread::sleep_for( std::chrono::microseconds( usec ) )```.

In practice this means the uPy executable will also depend on the C++ runtime instead of just the C runtime, which if I remember correctly adds 10 to 20kb of memory usage. And that there's C++ code in uPy. Nothing in particular is written about this in the guidelines except one can argue it is not 'micro'. So what are the opinions on this?

pfalcon, 2015-10-23:
> Sleep( usec / 1000 )

I'd say this is ok (I previously thought you mean that sub-second sleeps are not readily available in Windows). Bringing in C++ for this isn't worth it. (And besides not too faithful emulation, there's also option of not implementing sleep_us)( for Windows, though again, it's more chore than "sloppily emulating" it).

dhylands, 2015-10-23:
You may want to look at QueryPerformanceFrequency and QueryPerformanceCounter.  https://msdn.microsoft.com/en-us/library/windows/desktop/ms644905(v=vs.85).aspx https://msdn.microsoft.com/en-us/library/windows/desktop/ms644904(v=vs.85).aspx

stinos, 2015-10-24:
That is usable and accurate indeed, but only for busy waits. Using https://msdn.microsoft.com/en-us/library/windows/desktop/ms686289%28v=vs.85%29.aspx one can do a 'kernel wait' (or whatever correct terminology is) in 100 nanosecond increments.

dhylands, 2015-10-24:
Ah yes - that would probably be better. When I first looked at that I saw milliseconds for the period and missed the 100-nsec part.

stinos, 2015-10-25:
Closed with #1536

closed 2015-10-25 by stinos