micropython/micropython #1532: Windows port does not build in Visual Studio 2015

omtinez, 2015-10-25:
I am trying to get a Micropython binary for Windows, and there are many issues with the existing port. I have made some changes that allow me to build the project, but linking fails miserably. Unfortunately, this is way out of my comfort zone so I am not able to figure out what is failing at the linking step.

My changes are summarized here: https://github.com/omtinez/micropython/commit/9ac7a21b6dc70b436064b9d5d5486b3cd08aaf21.

Some things worth noting:
- lwIP was missing some key components regarding the OS specific implementation. I am baffled about how this project could possibly build in the past without those. I added them straight out of the `lwip-contrib` repo into `/micropython/windows/lwip-win32`
- All references to `ip_addr` in lwIP are completely broken, since they all point to `ipv4/lwip` instead of directly `lwip`. I had to do a mass string-replace to get the building to work, and that is not reflected as part of the commit linked above since it's in a different repository (where does exactly the lib folder come from to get populated by the `git submodule update --init` command?
- One of the changes required for the project to build affects the unix port, which I am unable to test at the moment. Special care needs to be taken to that to ensure that the unix port is unaffected by this change.

The output of the linker is the following:

	Build FAILED.

	"D:\git\micropython\windows\micropython.vcxproj" (default target) (1) ->
	(Link target) ->
	  modlwip.obj : error LNK2019: unresolved external symbol _netutils_format_inet_addr referenced in function _lwip_socke
	t_recvfrom [D:\git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _netutils_parse_inet_addr referenced in function _lwip_socket
	_connect [D:\git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _lwip_init referenced in function _mod_lwip_reset [D:\git\mic
	ropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _pbuf_alloc referenced in function _lwip_udp_send [D:\git\mic
	ropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _pbuf_free referenced in function _lwip_socket_close [D:\git\
	micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _pbuf_copy_partial referenced in function _lwip_tcp_receive [
	D:\git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_new referenced in function _lwip_socket_make_new [D:\git
	\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_arg referenced in function _lwip_socket_make_new [D:\git
	\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_accept referenced in function _lwip_socket_listen [D:\gi
	t\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_recv referenced in function _lwip_socket_connect [D:\git
	\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_err referenced in function _lwip_socket_make_new [D:\git
	\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_recved referenced in function _lwip_tcp_receive [D:\git\
	micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_bind referenced in function _lwip_socket_bind [D:\git\mi
	cropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_connect referenced in function _lwip_socket_connect [D:\
	git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_listen_with_backlog referenced in function _lwip_socket_
	listen [D:\git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_abort referenced in function _lwip_socket_close [D:\git\
	micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_close referenced in function _lwip_socket_close [D:\git\
	micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _tcp_write referenced in function _lwip_tcp_send [D:\git\micr
	opython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_new referenced in function _lwip_socket_make_new [D:\git
	\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_remove referenced in function _lwip_socket_close [D:\git
	\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_bind referenced in function _lwip_socket_bind [D:\git\mi
	cropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_connect referenced in function _lwip_socket_connect [D:\
	git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_recv referenced in function _lwip_socket_make_new [D:\gi
	t\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_sendto referenced in function _lwip_udp_send [D:\git\mic
	ropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _udp_send referenced in function _lwip_udp_send [D:\git\micro
	python\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _dns_gethostbyname referenced in function _lwip_getaddrinfo [
	D:\git\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _HAL_Delay referenced in function _lwip_socket_connect [D:\gi
	t\micropython\windows\micropython.vcxproj]
	  modlwip.obj : error LNK2019: unresolved external symbol _sys_check_timeouts referenced in function _mod_lwip_callback
	 [D:\git\micropython\windows\micropython.vcxproj]
	  modtime.obj : error LNK2019: unresolved external symbol _usleep referenced in function _mod_time_sleep_us [D:\git\mic
	ropython\windows\micropython.vcxproj]
	  D:\git\micropython\windows\micropython.exe : fatal error LNK1120: 29 unresolved externals [D:\git\micropython\windows
	\micropython.vcxproj]

		0 Warning(s)
		30 Error(s)

pfalcon, 2015-10-25:
> lwIP was missing some key components regarding the OS specific implementation. I am baffled about how this project could possibly build in the past without those. 

It never never did, not it's supposed to.

@stinos : Please remove any star-includes from MSVC project files.

stinos, 2015-10-25:
see #1535

pfalcon, 2015-10-25:
@stinos : Thanks. Can you also do something about this: https://github.com/micropython/micropython/pull/1513#issuecomment-148940671 , as this same issue appears to be tackled in the patch above too.

omtinez, 2015-10-25:
Excellent, now I'm able to build without using any of the /lib components. I cleaned up the necessary changes for the build in Visual Studio 2015 to work and opened a pull request: https://github.com/micropython/micropython/pull/1538

pfalcon, 2015-11-16:
Assuming fixed.

closed 2015-11-16 by pfalcon