micropython/micropython #1544: Branches cleanup

pfalcon, 2015-10-26:
I'd like to propose creation of a new branch, and so would like to make sure we don't have too many branches ;-).

So, I believe "osx" is completely stale - osx build should be well supported in master.

Then we have "dynamic-native-modules" and "loadable-native". It's unclear how they differ, and rather be integrated into one branch, and I'd say actually better merge it into master, even if disabled by default, to give people better chance to play with it.

dpgeorge, 2015-10-28:
I agree we can get rid of pretty much all branches.  I now put any PRs of my own on a branch at my personal repo (dpgeorge/micropython) (should have done this from the start...) so we can delete all those old PR ones as well.

pfalcon, 2015-10-28:
> I now put any PRs of my own on a branch at my personal repo (dpgeorge/micropython) (should have done this from the start...) so we can delete all those old PR ones as well.

Sorry, what I'm trying to do is exactly to go in the opposite direction - to propose to move branches which I had in my personal repo, but should be useful for wider audience, into the main repo.

So, I'd suggest against move your branches away were people (me too) won't even have a chance to stumble upon. What I propose is just to do quick cleanup of branches not really needed.

pfalcon, 2015-10-29:
Here's a specific proposal for a new branch: https://github.com/micropython/micropython/issues/1549

dpgeorge, 2015-10-29:
Out of the stale branches, we can remove both osx and extint-overhaul (now implemented in new machine API).

I agree it'd be nice to merge dynamic loadable native modules into master, even if it's WIP (at least it actually works).  Then we can remove those 2 branches.

travis-testing is useful to keep and should be rebased to master.

pfalcon, 2015-10-29:
Ok, deleted "osx", not going to touch anything else, please do something about it when it'll be right time for you.

dpgeorge, 2015-10-29:
extint-overhaul has been deleted.

dpgeorge, 2015-10-30:
loadable-native has been deleted.  dynamic-native-modules is the branch containing the latest patches for this feature.  It has been rebased and still works!

pfalcon, 2015-10-30:
Thanks, I guess this can be closed then.

closed 2015-10-30 by pfalcon

dpgeorge, 2015-11-17:
I added a branch "parse-bytecode" which is used by microbit.  This branch aims to reduce the memory usage of the parser by encoding the parse tree as "parse bytecode" instead of a traditional multi-node tree with pointers linking up the nodes (which is the scheme the master branch uses).  parse-bytecode cannot be merged into master because it completely replaces the parser and compiler (it was a lot of work to implement it).  But it does reduce RAM usage by about 40% which is significant for microbit with around 10k heap, and I thought it might be useful for other constrained ports.  It adds about 2k to the code size and is slower to compile (but not by much).