micropython/micropython #1546: RTC LSE/LSI selection

peterhinch, 2015-10-28:
The patches offered by @chuckbook aimed to solve two problems. The first was an issue with oscillator initialisation, now subject to a PR. The second, by my (incomplete) understanding of the code, aimed to select the LSE or LSI oscillator at runtime. The aim of this issue is to solicit comments on the relevance and design of this proposed feature, notably from @chuckbook.

The LSI oscillator is a low precision RC oscillator internal to the chip, while the LSE is an oscillator requiring external components to achieve substantially improved accuracy. The LSE is implemented with a 32.768KHz crystal on the Pyboard.

To initiate the discussion I'd like to raise the following questions.

1. Does the LSI have merit other than in extremely cost constrained systems? Is it relevant to actual or proposed MicroPython target boards? Or does it have other relevance e.g. fault tolerance?

2. If it is relevant, is there a reason for performing the selection at runtime, rather than making it a target specific compile time option?

3. If done at runtime, when should selection occur? At initial power up? On soft boot or recovery from standby? Or must the choice be iterated or event driven and, if so, in response to what events?

4. If hardware containing an LSE can fall back to the LSI at runtime, from the user perspective this is a fault condition. I would submit that this should raise an exception.

chuckbook, 2015-10-28:
@peterhinch thanks for starting the discussion! Let me try to answer with respect to three possible scenarios.
a. Pyboard related
b. stmhal related
c. Micropython related
ad 1a
The Pyboard has a reasonable stable LSE (which is not very precise) The LSE/crystal combination provides very low power usage (compared to other SOC's). In general there is no need to even think about the LSI option. However, if the LSI option is selected at compile time, the standby/wakup has to be supported as well. This is not the case with the latest rtc.c patch mentioned above.
ad 1b
Other STM32 processors are sometimes very difficult to handle when it come to LSE. Actually the reason we started the LSE/LSI thing was when a new charge of chips suddenly did not start the LSE at all. In such a situation (at initial power on) we decided to fall back to LSI and keep thaf till the next (VBAT) power up. Of course timekeeping is lousy but sufficient for standby/wakeup.
ad 1c
It would be nice, if the general Micropython environment could provide information about state and quality of the underlying RTC subsystem.

ad 2 see above

ad 3
The LSE check has to be performed at initial (VBAT)power up as well as after recovery from standby. It may be useful to check for a valid RTC subsystem before entering wake-up/standby. Everything else seems to be too application specific.

ad 4
Monitoring RTC validity at run-time is a good thing per se but it should not be mixed with the general LSE/LSI handling.

For points 3 & 4 we decided to follow a different strategy. The basic idea is to transfer RTC time to an internal counter on start-up and eventually restore to RTC on standby. Of course this needs more modifications in mpy, but it is an interesting approach.

peterhinch, 2015-10-28:
Thanks for that, very informative. I'd still like some more background on the hardware issue.
*a new charge of chips suddenly did not start the LSE at all.*
Could you please clarify this? Did an existing hardware design suddenly fail when built with a new batch of chips? Or did a new design of board for a different STM variant fail? Have you encountered a situation where on a given board the oscillator sometimes starts, and sometimes fails?

I take it we aren't dealing with a hardware design issue of the value of the crystal loading capacitors?

Presumably a board whose crystal oscillator consistently failed to start would be rejected in production test.

chuckbook, 2015-10-28:
@peterhinch we had all the symptoms you mentioned (STM32F405 same board new charge, RTC start-up fail below -30°C, STM32F427 --> STM32F746 RTC start-up fail even at room temp). Look at the pyboard, using a 10ppm crystal and proper capacitors freq. deviation is sometimes more than 100ppm. We never observed such behavior with TI or Microchip MCU's. On the other hand I like to power a RTC and 4KByte of backup RAM with ~1µA at no extra cost.