micropython/micropython #1549: RFC: Move pfalcon's "cross" branch to mainline

pfalcon, 2015-10-29:
I maintain for a year or so "cross" branch in my fork - https://github.com/pfalcon/micropython/tree/cross . It contains build and test wrappers for various embedded platforms - e.g. MIPS little- and big-endian, recently Android was added. I can't claim it all done in the ideal way, so don't propose to merge it to master. But it's definitely useful, and worth being accessible to wider audience, so I propose to push this branch to the main repo. Intended mode of operation for that branch is to be rebased regularly on top of master.

dpgeorge, 2015-10-29:
+1 for making it a branch.

> Intended mode of operation for that branch is to be rebased regularly on top of master.

Yes, that sounds like the best option.  Otherwise it can become a real pain when you want to eventually merge it to master.  Just need to make it known that (all?) branches will be rebased regularly.  Remind me: if you do "git pull --rebase" then that takes care of the fact the remote branch may have been rebased?

pfalcon, 2015-10-29:
> +1 for making it a branch.

Thanks!

>  Remind me: if you do "git pull --rebase" then that takes care of the fact the remote branch may have been rebased?

Yes, that takes care of updating to even rebased upstream branches, while preserving local commits. Also, I have a feeling that newer version of git will detect rebases of tracked pristine upstream branches (i.e. checked out locally, but with no local commits) even without that (at least I don't use pull --rebase all the time with other projects, and sometimes get a "forced update" message for local branch). Well, that yet needs to be tested in clean conditions, but git pull --rebase definitely works ok.

pfalcon, 2015-10-29:
Ok, the branch is pushed to mainline and described at https://github.com/micropython/micropython/wiki/CrossBranch

pfalcon, 2015-10-29:
Ok, I did a fresh clone, which thus has:

~~~~
$ git br
* master
~~~~

Then pushed a commit to master, rebased cross and run git pull:

~~~~
$ git pull
remote: Counting objects: 14, done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 14 (delta 5), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (14/14), done.
From https://github.com/micropython/micropython
   ebd9f55..04fa999  master     -> origin/master
 + 8c5fe0c...4b272fd cross      -> origin/cross  (forced update)
Updating ebd9f55..04fa999
Fast-forward
 stmhal/pyexec.c | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)
~~~~

That's the behavior I meant above: if a branch which had been rebased isn't checked out locally, running just git pull is safe. So we now confirmed that for people who don't care about cross, rebasing it won't give them a hassle.

pfalcon, 2015-10-29:
Now I checkout cross locally (so its setup as a tracking branch), but switched back to master, so now I have:

~~~~
$ git br
  cross
* master
~~~~

And:

~~~~
$ git pull
remote: Counting objects: 16, done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 16 (delta 6), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (16/16), done.
From https://github.com/micropython/micropython
   04fa999..6a09e7d  master     -> origin/master
 + 4b272fd...6742182 cross      -> origin/cross  (forced update)
Updating 04fa999..6a09e7d
Fast-forward
 esp8266/esp_mphal.c | 2 +-
 esp8266/esp_mphal.h | 2 +-
 esp8266/modpyb.c    | 4 ++--
 3 files changed, 4 insertions(+), 4 deletions(-)
~~~~

Still good. So, we're pretty safe - rebasing may affect only people who work with the cross branch, and they should know about git pull --rebase.

closed 2015-10-29 by pfalcon

pfalcon, 2015-10-29:
~~~~
$ git --version
git version 2.4.3
~~~~

dpgeorge, 2015-10-29:
Sounds good, rebasing is the way to go!