micropython/micropython #1552: teensy: Remove adhoc memzip support

pfalcon, 2015-10-29:
@dhylands : If you agree that frozen modules is the way to go, can you please remove memzip code from teensy port to ease maintenance? If you don't have time, I can give it a go. Thanks.

dhylands, 2015-10-29:
Sure - I can take a look at that, and add back in the equivalent frozen module support (bundling up some python files to be included in the firmware).

dpgeorge, 2015-10-29:
Isn't memzip to create a full RO filesystem, not just restricted to Python scripts?  Eg I can put a text file in the memzip dir and then I can open it as usual and read the contents?  (Hmm, it seems that this is not implemented, but in principle it could...)

pfalcon, 2015-10-29:
The same functionality is needed for frozen modules, except per Python API, they should be accessed using https://github.com/micropython/micropython-lib/tree/master/pkg_resources & https://github.com/micropython/micropython-lib/tree/master/pkgutil (yeah, dudes couldn't even come up to a single module which does the thing). And I actually even worked on making them work for frozen modules, but now don't even remember the outcome...

Bottom line: this stuff should come from the other side - not "how to resolve particular port's problems", but "how to implement standard Python API in MicroPython (efficiently)".

dpgeorge, 2015-10-30:
Can you hook pkgutil and pkg_resources into the filesystem so `f = open('test.tx', 'r')` works?

Anyway, I'm happy to remove memzip if @dhylands thinks it's no longer useful/needed.

dhylands, 2015-10-30:
So yes - memzip is providing essentially what amounts to a RO file system to allow arbitrary files to be opened.

I didn't quite get how this was a maintennce issue.

Anyways, frozen modules will provide most, if not all, of the functionality that memzip provided. So if we can get pkgutil hooked up then I'm happy to remove memzip functionality (but preferrably not in the other order).

pfalcon, 2015-10-30:
> Can you hook pkgutil and pkg_resources into the filesystem so f = open('test.tx', 'r') works?

No, you can't. In Python, you can't either. Python requires that if you want to access resources in packages regardless of package format, you should use pkgutil and pkg_resources, not open().

The logic is hopefully clear - before going to maintain/develop non-standard solution, why don't we make standard first? (I don't say we should, just asking to consider.)

And then if memzip is really more like RO filesystem, it should be moved to be port-independent. That suggestion was sounded more than year ago when memzip was initially, and there're no changes on that frontend, so I'm just checking if there can be changes in other direction. If not - well, ... ok.

pfalcon, 2015-10-30:
> I didn't quite get how this was a maintennce issue.

It wasn't, but may become anytime. So, after https://github.com/micropython/micropython/commit/eb099b9893e896300241050a2c3ed272c3890d70 , and remembering that you yourself suggested frozen modules over memzip, I just wondered if we can minimize port delta which certainly will help maintenance. If not, and memzip still offers capabilities beyond frozen modules - sounds good, please just close this bug.

dpgeorge, 2015-11-04:
Addressed by #1569.

closed 2015-11-04 by dpgeorge