micropython/micropython #1553: Move pyexec.c from stmhal to where?

pfalcon, 2015-10-29:
https://github.com/micropython/micropython/commit/04fa999cfe86e731d03305081e3df42e9b60508b broke bunch of ports and it's not really big surprise, as commit message says. So, its status as cross-port should be confirmed by move away from stmhal. The question is where to.

Let's consider that top-level dichotomy we have is baremetal vs POSIX-like ports. The file in question is not a generic uPy file, its baremetal port file. So, we can't put it to py/, and extmod/ or lib/ don't look too good places either. The best idea I have so far is to setup lib/port/ for such stuff, to avoid another top-level dir. Thoughts?

dpgeorge, 2015-10-29:
pyexec.c could almost be used by unix port, since a lot of stuff is now duplicated there, eg friendly REPL: https://github.com/micropython/micropython/blob/master/unix/main.c#L163-L232

To me it almost feels like it should be part of py/repl.c, but since it relies on stuff from lib/ (readline) it shouldn't really go in py/ directory.  lib/port/ is a bit generic, since lib/README.md states that code in lib/ can be used by any port, so lib/ already is the same thing as lib/port/.

What else fits together with pyexec.c?  A lot of baremetal ports use stmhal/printf.c and stmhal/pybstdio.c.  Also stmhal/lexerfatfs.c and stmhal/file.c are relatively generic.

Maybe lib/bareio?

pfalcon, 2015-10-29:
> so lib/ already is the same thing as lib/port/.

Not quite, lib/ are generic libs not tied with the rest of uPy (well, maybe minimal dependencies to mp_hal_*).

pfalcon, 2015-10-29:
> pyexec.c could almost be used by unix port,

Yeah-yeah, I'm with that thought for at least half a year, and nothing changes. That's why I draw up that multi-level hierarchy - let's first resolve at least duplication within baremetal ports, then think unix vs baremetal.

dpgeorge, 2015-10-30:
> Not quite, lib/ are generic libs not tied with the rest of uPy

mp-readline, netutils and timeutils are all dependent on main uPy headers (mpstate.h and/or obj.h).

pfalcon, 2015-10-30:
> mp-readline, netutils and timeutils are all dependent on main uPy headers (mpstate.h and/or obj.h).

Ok, I'm still not sure that lib/bareio is best choice, because as pointed above, it may hopefully be reused by unix port, and limiting it to "io" also probably too narrow.

dpgeorge, 2015-10-30:
How about lib/utils or lib/ioutils?  Similar to existing netutils and timeutils.  pyexec.c, pybstdio.c, printf.c, file.c and lexerfatfs.c all deal with io so ioutils makes sense.

pfalcon, 2015-10-30:
I'd still prefer to avoid "ioutils" - we don't know what may end up there later, so "ioutils" may be awkward. lib/utils sounds good, I'll proceed then.

dpgeorge, 2015-10-30:
Maybe stmhal/printf.c can go in lib/libc?

pfalcon, 2015-10-30:
That, or otherwise go to lib/utils/python_printf, if it's actually not standard printf, but something with upy-specific behavior.

dpgeorge, 2015-10-30:
stmhal/printf.c is actually quite generic.  Only thing is that it's backend is mp_vprintf, which is Py specific.

pfalcon, 2015-10-31:
Started refactoring with 0ec51441de4db3fc776b1f2013005391caa910d3

dpgeorge, 2015-11-10:
pyexec.h header moved to lib/utils in 40274fec9c0f8e1ce06f999927f550106b9aa5e8.

pfalcon, 2015-11-10:
printf.c moved too, commit message explains why to utils/ after all: https://github.com/micropython/micropython/commit/73ff0687f2f5331e962a4af0126097e9921db631

pfalcon, 2015-11-10:
Ok, next in queue is pybstdio.c. Mmm, not sure where it can be moved. What it does is that it's not implementing entire module, just some Python objects, which will be collected into modules elsewhere. And it implements them in terms of mp_hal_*(). This is the same pattern for sleep_us/ms/etc. refactoring which I have in mind. So, both files would need to be put at the same place. And for sleep_*, I was actually thinking about extmod/.

Any ideas?

dpgeorge, 2015-11-12:
I was thinking to rename pybstdio.c to mpstdio.c.  Could it even go in py/?  Could it be refactored a bit and used to replace the mp_sys_std{in,out,err} stuff in unix/file.c?

pfalcon, 2015-11-12:
> Could it even go in py/?

The we'll need to re-assess criteria for what goes into py/ (core), because so far i/o implementation wasn't core.

> Could it be refactored a bit and used to replace the mp_sys_std{in,out,err} stuff in unix/file.c?

If it could, would be nice, but I didn't think so far. From quick look I did at the time of writing, https://github.com/micropython/micropython/issues/1553#issuecomment-155603157 ,  unix stdio was based on OS fd's, whereas baremetal on emulated fd's and mphal.

dpgeorge, 2015-11-12:
>> Could it even go in py/?
>
> The we'll need to re-assess criteria for what goes into py/ (core),
because so far i/o implementation wasn't core.

But sys module is in core so it's borderline decision to put
sys.stdin/out/err there as well.

>> Could it be refactored a bit and used to replace the
mp_sys_std{in,out,err} stuff in unix/file.c?
>
> If it could, would be nice, but I didn't think so far. From quick look I
did at the time of writing, #1553 (comment) , unix stdio was based on OS
fd's, whereas baremetal on emulated fd's and mphal.

Yes there are differences in implementation so there would need to be some
refactoring/port-configulability. But might be worth it so each port
doesn't have to think how to do sys.stdio correctly each time. Especially
if we want overrideable attrs and sys.stdin.buffer etc support.

pfalcon, 2015-11-13:
Well, I'd say that sleep_us/etc is more low-hanging fruit. Do you agree that those should go into extmod/ (not complete modules, just Python function objects)?

dpgeorge, 2015-11-13:
> Well, I'd say that sleep_us/etc is more low-hanging fruit. Do you agree that those should go into extmod/ (not complete modules, just Python function objects)?

Yes, sounds sensible.  If they go in extmod/modutime.c then it would be different from the other extmod/mod*.c files in that it provides just the functions, not module wrapper.

Issue is that stmhal's sleep_us/etc check for positive argument, so need a way to support that.

pfalcon, 2015-11-13:
> If they go in extmod/modutime.c then it would be different from the other extmod/mod*.c files in that it provides just the functions, not module wrapper.

Yes, that's what I mean above. So, it wouldn't be called modutime.c, but rather e.g. utime_funcs.c

> Issue is that stmhal's sleep_us/etc check for positive argument, so need a way to support that.

Sure, I'm ready to give up unix "special behavior" in the name of no code duplication. I'd still suggest to keep documenting it as undefined, to leave freedom for super-optimization for viper/JIT.

dpgeorge, 2015-11-13:
> So, it wouldn't be called modutime.c, but rather e.g. utime_funcs.c

Yes, that's ok.

Note that we'll need to add sleep_us and ticks_us to py/mphal.h...

pfalcon, 2015-11-21:
Ok, mp_verbose_flag issue gets resolved in 295ea12411ca12a41947de3113febb29e469ac4c

pfalcon, 2015-11-21:
One the things which pulls in stdio is assert(). Should we provide our own implementation?

dpgeorge, 2015-11-21:
> One the things which pulls in stdio is assert(). Should we provide our own implementation?

stmhal does have it's own assert: https://github.com/micropython/micropython/blob/master/stmhal/main.c#L109-L115

Thing that's not portable there is what to do to abort the process?

pfalcon, 2015-11-21:
Well, even if it's implemented per-port, it's good enough, so I'll look into that.

In the meantime, I switch unix to use utils/printf.c : https://github.com/micropython/micropython/issues/1553#issuecomment-158688358 . That magically not breaks glibc dynalinked build (i.e. glibc still manages to override snprintf() in utils/printf.c - I din't bother to figure why, likely because snprintf is defined by headers to __GI_snprintf).

But it breaks static build, but portable double formatting code is in works (yes, I decided to go that way, because trying hard to get rid of stdio just to then try hard to find a loophole to pull libc's snprintf is just too depressing).

pfalcon, 2015-11-21:
> That magically not breaks glibc dynalinked build

Well, that broke coverage build, hitting assert protecting against infinite recursion...

pfalcon, 2016-06-19:
Closing for now, bunch of refactors for that was done.

closed 2016-06-19 by pfalcon