micropython/micropython #1556: Error in inline assembler documentation

peterhinch, 2015-10-30:
The documentation for the ``vldr`` and ``vstr`` instructions is incorrect: offsets are measured in 32 bit words, not bytes.

This came up in review: the manual clearly states that offsets are in bytes and I "corrected" the text accordingly. Alas this seems to be a typo in the manual (not the first). This seems definitive:
```python
from array import array
test = array('i',(0x12345678, 0x33333333, 0x7654321))
@micropython.asm_thumb
def bar(r0):
    vldr(s0, [r0, 1])
    vmov(r0, s0)

@micropython.asm_thumb
def foo(r0):
    vldr(s0, [r0, 2])
    vmov(r0, s0)

print(hex(bar(test)))
print(hex(foo(test)))
```
The outcome:
```
MicroPython v1.4.6-117-gee9b02f-dirty on 2015-10-16; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> import rats5
0x33333333
0x7654321
>>> 
```
We're emitting the offset verbatim in ``emitinlinethumb.c`` so the error is in the manual.

dpgeorge, 2015-10-30:
I think it depends on how the assembler works: does it divide the offset by the word size (4 in this case) or not?

With ldr/str we divide the offset by 4, see https://github.com/micropython/micropython/blob/master/py/emitinlinethumb.c#L708 . This is consistent with how ARM assembler is written, which is what we aim for.

So, if the ARM manual says vldr/vstr is measured in bytes, then that means the ARM assembler accepts bytes and divides the offset by 4 before encoding.  And we should do the same.

peterhinch, 2015-10-30:
OK, that figures. The manual says:

"<imm> The immediate offset used to form the address. [...] Permitted values are multiples
of 4 in the range 0 to 1020."

The field width for the offset is 8 bits implying that the assembler performs the right shift. If we are to comply with this our docs are correct and the code needs modifying. I can submit a PR if we're in agreement on this.

dpgeorge, 2015-10-30:
> The field width for the offset is 8 bits implying that the assembler performs the right shift. If we are to comply with this our docs are correct and the code needs modifying. I can submit a PR if we're in agreement on this.

Yes please.

dpgeorge, 2015-10-31:
FIxed.

closed 2015-10-31 by dpgeorge