micropython/micropython #1559: stmhal: Improve RTC startup

dpgeorge, 2015-10-30:
We can improve the way stmhal starts up and initialises the RTC.  This is tied into the way that the device resets, either by power up, hard reset, soft reset, or exiting standby mode.  Some desired features:

- Detect exactly when RTC initialisation is needed, and when calendar reset is need.  This may already be solved but needs to be tidied up.

- Support LSE/LSI detection on startup, see #1546.

- Perform LSE startup "in the background" while booting, because currently it can take up to 500ms.

- Provide a way to indicate to the user what the reset cause was.  machine.reset_cause() and machine.wake_reason() are used for this.

- It would be good to set the CPU clock to it's previous value when exiting standby mode, since in low power applications you wan't a slower clock and having it 168MHz on boot (even for a few ms) wastes power.  This is not really RTC related, but is related to detecting what kind of reset occurred.

chuckbook, 2015-10-30:
Just completed a patch that includes the LSE/LSI handling with initial fallback on faulty LSE.
I'm not sure if "background startup" makes sense.
If we agree upon a dedicated memory location in the backup area, we have a solution:
valid LSE/LSI --> valid (freq-)backup --> defined frequency

TDAbboud, 2015-11-02:
I am currently working on implementing bullets 1 and 4, but I have a question regarding the reset states. What is the difference between the power-up and hard reset? I am under the impression that a power-up reset is a type of hard reset, but a hard reset does not necessarily have to be a power-up reset. How else can you perform a hard reset?

dhylands, 2015-11-02:
pyb.hard_reset() or machine.reset()

peterhinch, 2015-11-02:
@TDAbboud A few thoughts regarding bullet 4. The Pyboard can be woken from deep sleep by interrupts on the Tamper and WKUP pins (X18 and X1 respectively) in addition to an RTC wakeup event. This is how I distinguish them
```python
def why():
# after detecting and returning power up status
    rtc_isr = stm.mem32[stm.RTC + stm.RTC_ISR]
    if rtc_isr & 0x2000 == 0x2000:
        return 'TAMPER'
    if rtc_isr & 0x400 == 0x400:
        return 'WAKEUP'
    return 'X1'
```
I haven't found a way properly to detect WKUP events on X1 - so currently it's assumed in the absence of any other cause. Perhaps you know the answer to this?

dpgeorge, 2015-11-02:
I think point 1 is being solved by #1571.

TDAbboud, 2015-11-03:
@dpgeorge I didn't see that, so I will just continue with point 4 then.

After doing some testing on the reset states I have found that a Power-on hard reset differs slightly than the hard reset done through machine.reset(). The machine.reset sets the software reset (SFTRST) flag, which is not currently checked in the rtc code.

@peterhinch I have to do some more reading in the datasheet, but from what I have read you should be able to check the WUF (wakeup) flag which is bit 0 of the PWR_CSR register. There is also the WKUPINT in the USB_OTG registers

dhylands, 2015-11-03:
We need to code a reset_cause function which sounds like it could be shared with/used by this code.

peterhinch, 2015-11-03:
It would be good if reset_cause returned the pin which caused a pin wakeup. Perhaps by means which can be generalised for other platforms. This would save ``stm.mem32()`` hacks.

dpgeorge, 2016-12-02:
RTC startup was improved long ago, and feature to restore PLL values to previous state was also added.  reset_cause() method was added to machine module (but see #2500 for further improvements to that).

closed 2016-12-02 by dpgeorge