micropython/micropython #1566: How to regenerate documentation (to test new markup etc).

dhylands, 2015-10-31:
I'm trying to add some documentation and I'd like to regenrate it locally so that I can see if my markup is good.

I tried cd'ing into the docs directory and running make html, but it doesn't look anything like the online docs.

So I figured there is something simple I'm missing. I'll take whatever I find (from others posting here) and create a readme.txt file in the docs directory so that we'll have some documentation.

I did search through a bunch of stuff but obviously wasn't using the right keywords or looking in the right places.

pfalcon, 2015-10-31:
I told we have too long README.md, people don't read it ;-).

dhylands, 2015-10-31:
I have read it several times, but obviously not since that was added.

And it feels like the wrong place for that information (this is my
constructive criticism :)

I'd propose that we move that section from the topmost README.md and
instead put it in docs/README.md

If you're in agreement, I'd be happy to do that as part of my PR (I'll put
that in a separate commit)

pfalcon, 2015-10-31:
It was added a year ago less 5 days ;-).

> And it feels like the wrong place for that information (this is my constructive criticism :) 

That what I said to the PR which added it ;-)

dhylands, 2015-10-31:
My memory is lucky to go back a month let alone a year :)

ok - that worked (modulo a minor error in the documentation). I'll fix that when I do the commit to move it to the docs directory.

pfalcon, 2015-10-31:
A day ago first suggestion (docs for "unix" port ) didn't work at all, and there're actually >1 sphinx "warnings" here and there (when building for different ports). Docs definitely need love, thanks for looking into that, and +1 to move it to docs/README and re-read main README from time to time and see if it can be optimized further ;-).

dhylands, 2015-11-01:
Closing this, since I moved the documentaion in #1567

closed 2015-11-01 by dhylands