micropython/micropython #1578: minimal: Question about the Makefile

dhylands, 2015-11-02:
I noticed that in the minimal/Makefile that it has these lines:
```
SRC_S = \
#	startup_stm32f40xx.s \
	gchelper.s \
```
which means that gchelper is not included in SRC_S. The newline is escaped so gchelper.c is included in the comment. And when you view the Makefile in github, it actually shows both lines grayed out as comments.

There is also no gchelper.s in the minimal directory, so it may be deliberate.

This question is to determine what the correct intention is.

dpgeorge, 2015-11-02:
Correct intention is that gchelper.s is not included in the minimal build.

If you look in main.c at gc_collect, there's a warning that CPU registers are not scanned for root pointers.  That's exactly what gchelper.s would do, but it's not portable.

dhylands, 2015-11-02:
ok - I was going to comment out both lines then.

Does it perhaps make sense to have minimal configured to use the setjmp method by default (this should at least be portable) and then add a comment to explain what needs to be done to use gchelper/nlrXxxx

dpgeorge, 2015-11-02:
I'd err on the side of not changing anything in minimal.

pfalcon, 2015-11-02:
The original idea of minimal is being a port to measure code size, which can run, unlike bare-arm, which can't. It can be taken as a basis for other ports too. Changing it indeed may use consideration what is expected to be achieved. For example, I guess @dpgeorge suggested to use it with emscripten for *initial* experiments, if they go well, you may as well try bigger port next for example.

dhylands, 2015-11-02:
> For example, I guess @dpgeorge suggested to use it with emscripten for initial experiments, if they go well, you may as well try bigger port next for example.

I've created a seperate directory for emscripten which is a copy of minimal. I wasn't trying to modify minimal for emscripten, I just noticed some stuff when I was looking at the files.

pfalcon, 2015-11-07:
Ok, so original issue is now fixed, closing.

closed 2015-11-07 by pfalcon