micropython/micropython #1588: Running pyboard.py produces stray \x04 in the stdout output

pfalcon, 2015-11-05:
To reproduce:

~~~~
$ ./pyboard.py --device /dev/ttyACM0 -c "print(1)"
1
\x04
~~~~

Where "\x04" is a unicode character with that code. I don't know if that's expected, doesn't look to aesthetically.

dpgeorge, 2015-11-06:
Ack.  I saw this before but it didn't bother me (on my terminal I need to pipe the output into xxd to see the EOF char) so left it alone.

I think it's easy to fix and shouldn't break anything.

pfalcon, 2015-11-07:
> I think it's easy to fix and shouldn't break anything.

Well, it's easy to workaround: https://github.com/micropython/micropython/pull/1598 . Proper fix, like \x04 never gets to stdout_write_bytes(), would be more boring, and indeed, nothing warrants to do it like that.

pfalcon, 2015-11-07:
Fixed.

closed 2015-11-07 by pfalcon