micropython/micropython #1596: Stream objects are not checked for in some places, eg print()

dpgeorge, 2015-11-06:
The following crashes unix uPy:
```python
print(file=1)
```
Reason is that print() (or mp_stream_write, depending on how you look at it) doesn't validate its input as a stream.

blmorris, 2015-11-07:
I have a bit of a dumb question - I was assuming that this code section was to validate the input as a stream, and I was using it as such: https://github.com/micropython/micropython/blob/master/py/stream.c#L177-L182

I did notice from the coveralls report on #1581 (https://coveralls.io/builds/4035114/source?filename=py%2Fstream.c) that the invalid stream object case never gets called by the test suite.

pfalcon, 2015-11-07:
> I did notice from the coveralls report on #1581 (https://coveralls.io/builds/4035114/source?filename=py%2Fstream.c) that the invalid stream object case never gets called by the test suite.

And that's not good, we should test such paths too. But otherwise, the logic is as I wrote above - "no valid program should hit those paths", so it gets to the bottom of things someone thinks about when writes a test.

pfalcon, 2015-11-07:
I mean, we should test it, if we handle it. And how soon we handle that is another question...

pfalcon, 2015-11-07:
> doesn't validate its input as a stream.

Yes, and everything will validate everything, it will be +10% to the size of code or +10% to the execution overhead, which explains why people don't jump to it right away ;-).

blmorris, 2015-11-07:
> Yes, and everything will validate everything,

Except in this case we aren't talking about validating everything, we are talking about print / mp_stream_write, which is where I found this code. Does this mean that it's not an effective check?

dpgeorge, 2015-11-07:
> Does this mean that it's not an effective check?

It's an effective check.  But it doesn't check if the input (self_in) is even a concrete object (it could be a small int or qstr, and then it doesn't even have a ->type field).

But I don't think such a check is needed here, because in most cases mp_stream_write is passed a concrete type.  The check should go outside mp_stream_write (eg in mp_builtin_print).

blmorris, 2015-11-07:
> But it doesn't check if the input (self_in) is even a concrete object (it could be a small int or qstr, and then it doesn't even have a ->type field).

Thanks for the explanation, I get it now.

alex-robbins, 2017-07-10:
How does this issue relate to subclasses? For example, if you subclass `usocket.socket`, and pass it to `ussl.wrap_socket`, you currently get undefined behavior (it seems the passed in object is used as if it were a `usocket.socket`, not a subclass of `usocket.socket`). 

Is searching the inheritance hierarchy something that these built-in functions are supposed to do? If so, is that covered by this issue, or is it a separate one? 

(I'm trying to figure out if I should file a bug against micropython-lib about the fact that instances of `socket.socket`, which is a subclass of `usocket.socket`, cannot be passed to `ussl.wrap_socket`. It's clearly *a* problem, I just don't know if it's the same as this issue or if it should be fixed in micropython-lib or what.)

dpgeorge, 2017-07-11:
> How does this issue relate to subclasses?

Allowing a subclass of a native object as an argument, as well as just the native argument, is a separate issue and not easy to solve in the general case.  See for example #2749 for a particular case (and solution) for subclasses of dict.

> I'm trying to figure out if I should file a bug against micropython-lib about the fact that instances of socket.socket, which is a subclass of usocket.socket, cannot be passed to ussl.wrap_socket

That needs to be fixed in the core; again see #2749.

dpgeorge, 2018-06-20:
Checking for stream objects was streamlined in 6abede2ca9e221b6aefcaccbda0c89e367507df1 and e8398a58567cc94b866d46721fd06289601f5c8a, and a couple of additional checks (for print and sys.print_exception) added in 582b190764641e5049768da1ba8962c841ec014b.  That should now ensure that all stream objects are checked for, but only at "the beginning" of their use, and all subsequent stream use bypass any checks.

closed 2018-06-20 by dpgeorge