micropython/micropython #1602: stmhal: Add STM32F7DISC board to travis build

dhylands, 2015-11-07:
This is a request to add the STM32F7DISC board to the travis build.

pfalcon, 2015-11-07:
There's idea that we should be spare about adding stuff to Travis build (increases wait time). At the very least, I have some things to run in Travis which are more important than another random board (like, stackless build). And I don't propose to add them to mainline Travis (yet), instead running it in my fork's branch (rebasing from time to time). I don't see why you can't do the same.

dhylands, 2015-11-08:
I setup a branch called travis-master in my repository, and I setup a cron entry which runs on my local machine once per hour. Whenever it detects that master (in micropython/micropython) advances, then it rebases travis-master (in dhylands/micropython) to master and pushes it, which then triggers a build of the following targets:

ESPRUINO_PICO, CERB40, NETDUINO_PLUS_2, and STM32F7DISC

I'll get an email if any of those fail. I can check the status online at:
https://travis-ci.org/dhylands/micropython/builds/

If went from red to green when #1600 was merged.

closed 2015-11-08 by dhylands

dpgeorge, 2015-11-08:
Good interim solution.

When we upgrade the travis config to use their latest infrastructure then maybe it'll be fast enough that we can add more builds.

Otherwise we can keep travis lean and fast I can set up a dedicated build system that builds a lot more configurations/ports (already the docs and firmware are built automatically, so shouldn't be that hard).

dhylands, 2015-11-08:
> already the docs and firmware are built automatically

I haven't seen the repl docs updated yet. This page: http://docs.micropython.org/en/latest/reference/index.html still only shows the inline assembler. I haven't seen the REPL stuff show up.

dpgeorge, 2015-11-08:
Sorry, I should say that the docs are at http://micropython.org/resources/docs/en/latest/pyboard/ .  We need to redirect docs.micropython.org to this (ie move away from readthedocs servers).  But first we need a drop-down menu to select between ports, versions and languages.  Anyone up for that?

dhylands, 2015-11-08:
http://micropython.org/resources/docs/en/latest/pyboard/reference/index.html shows the same thing as http://docs.micropython.org/en/latest/reference/index.html

http://micropython.org/resources/docs/en/latest/pyboard/reference/index.html says it was last updated Oct 21.

http://docs.micropython.org/en/latest/reference/index.html refers to hash 126373ac which is also Oct 21.

dpgeorge, 2015-11-08:
You're right... there was an error in the crontab which is now fixed.

dpgeorge, 2015-11-08:
@pfalcon @dhylands I tried getting new Travis infrastructure running.  I could get everything working (even coveralls) except for installing arm-none-eabi-gcc, which is a real problem :(  Apparently it's a known issue that such a package can't be installed; see https://github.com/travis-ci/travis-ci/issues/4838 

See https://travis-ci.org/micropython/micropython/builds/90002575 for the successful Travis run on their new systems.  It builds minimal, unix and windows ports, and runs full coverage testing.  It takes 2 min, 39 sec.  Compare this with existing times of between 5:30 and 7 mins.  Difficult to compare since existing Travis runs build a lot more.

See .travis.yml file in travis-testing branch if you're interested in the config (it's pretty easy)