micropython/micropython #1607: makeing gc_free a NOOP has no effect on tests

vitiral, 2015-11-08:
Something that needs to be mentioned clearly is that, currently, replacing `gc_free` with a `NOOP` (by just putting a `return;` at the top) **does not affect the tests** -- all tests still run!

I would say that this code path is not adequately tested. I have [tried to write a test to address this](https://github.com/vitiral/micropython/blob/mem_pluggable/tests/basics/gc_collect.py), and no matter what values I put into there I cannot get the system to fail when `gc_free` is a `NOOP`. I don't really understand how this is possible, does anybody have insight?

dhylands, 2015-11-08:
That's because the garbage collector doesn't call gc_free to release collected memory. Calling gc_free is really just an optimization. It allows the calling code to say that it's definitely done with a given chunk of memory, which essentially means that the first call to the garbage collector can be deferred, since the memory that was freed will be available to be reallocated.

vitiral, 2015-11-08:
Ah, I gotcha. You would think that this would be done in `gc_sweep`, but actually it is done inline there without making a call to `gc_free`.

Thanks!

closed 2015-11-08 by vitiral

dpgeorge, 2015-11-08:
And as @pfalcon said somewhere once in a comment: one option to reduce the size of a build would be to turn m_free/gc_free into an inlined-nop since it is, as @dhylands' points out, an optimisation.