micropython/micropython #1610: def allows keyword arguments after a ** without SyntaxError and messes behavior up

gpshead, 2015-11-09:
MicroPython v1.5-1-ge954604 on 2015-10-21; WiPy with CC3200

```python
>>> def A(x, *a, **kw, r=3):
...     print(x, a, kw, r)
... 
>>> A(1,2,e=5,r=9)
1 {'e': 5} (2,) 9
```

By adding a keyword argument after the **kw in the above example, the behavior of the function is messed up.  a is assigned to the keyword dict, kw is assigned to the args tuple (and who knows what else might be going wrong internally...)

Accepting keyword arguments in a def after the ** should be a SyntaxError. (confirmed in Python 3.4 and 3.5)

dpgeorge, 2015-11-09:
Yes, this was known since the beginning to be an open problem, see https://github.com/micropython/micropython/blob/master/py/grammar.h#L310

It needs a bit of code in the compiler to check for args after **.

dpgeorge, 2015-11-23:
Fixed by 9a56912ad16065c8fc3670c8d493f922bc54e5b1.

closed 2015-11-23 by dpgeorge