micropython/micropython #1616: Building on system with only python3 fails

jhdulaney, 2015-11-12:
Building on a system with only python3 available fails:

python ../py/makeqstrdata.py build/genhdr/qstrdefs.preprocessed.h > build/genhdr/qstrdefs.generated.h
/bin/sh: python: command not found

dhylands, 2015-11-12:
I did a bit of investigation, and it seems like the only way to fix this is do something like the following:
```
sudo update-alternatives --config python
```
and select python3. If that doesn't work, then you can try:
```
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10
```

The above will make /usr/bin/python point to python3. I'm not sure how else to make the scripts work in all of the following environments:
1 - python 2 only installed
2 - python 2 and python 3 both installed
3 - python 3 only installed

jhdulaney, 2015-11-12:
Unfortunately, this is not possible in a distributions build system.

dhylands, 2015-11-12:
Another solution is to create a local directory which contains a symlink:
```
mkdir ~/bin
ln -s /usr/bin/python3 python
export PATH=~/bin:${PATH}
```
and then perform the build.

jhdulaney, 2015-11-13:
Once again, this will not work with distrubution build environments.

I point you to pep 394, and that you should explicitly call python2 or python3

dhylands, 2015-11-13:
> Once again, this will not work with distrubution build environments.

Pardon me for being ignorant, but what exactly is a "distrubution build environment" (I can guess - but I'd rather that you be explicit) and why is it important? What's the context here? Why is it that you're using one of these?

> I point you to pep 394, and that you should explicitly call python2 or python3

We support both, so limiting our support to one or the other would break the build for somebody. So this becomes a @dpgeorge decision...

I suppose we could always check (in the Makefile) to see if python3 runs, and if it doesn't then try python (or do the reverse order - whichever the preference is).

The other option would be to write a configure script which checks which version of python is installed and have the Makefile be generated (which is probably what most packages do - I'm just throwing this out as a suggestion that would work - but I'm not expecting that @dpgeorge would be in favor of it).

jhdulaney, 2015-11-13:
If you are officially supporting python2 and python3, then the build should not fail on the presence or absence of either; if it does and you do officially support python2 and python3, then it is a bug.  If you do not officially support both, then state it, and ensure that your documentation does not say otherwise, once again a bug.

There is no distribution around that is going to run commands as root during the build phase of a package, nor will they install alternative pythons in a second directory.  Asking that they do so rather than fixing what is clearly a bug of some sort, either in your build scripts or in your documentation, will not fly.

dhylands, 2015-11-13:
Reading PEP 394, it says you should use python (and not python2 or python3) if your script is compatible with both, and that is exactly what we do.

jhdulaney, 2015-11-13:
But, it should not fail in the way it does.

dhylands, 2015-11-13:
How should it fail then?

pfalcon, 2015-11-13:
Common sense says that if Python3 is the only available version, then it should be available as "python". I would be ready to close this ticket.

jhdulaney, 2015-11-13:
That's not what the pep says.  In fact, the pep explicitly states otherwise.

It should not fail; if it is capable of building with py3 and python3 is present, then it should continue.  If it is not capable of building with python3, then remove that from your documentation.

dhylands, 2015-11-13:
This is quoted from https://www.python.org/dev/peps/pep-0394/ (in the recommendation section):

"One exception to this is scripts that are deliberately written to be source compatible with both Python 2.x and 3.x. Such scripts may continue to use python on their shebang line without affecting their portability. "

drohm, 2015-11-13:
On Windows 10, I only have Python 3 installed.  If I open a console and type `python`, Python 3.5.0 launches.  On my laptop I'm running Xubuntu 15.10.  If I type `python`, Python 2.7.10 launches.  If I type `python3`, Python 3.5.0 launches.  Re-configuring your system to launch Python 3.x by typing python is actual discouraged as it could bork up your system, or so I've heard.

jhdulaney, 2015-11-13:
The way most distributions are handling Python 2 and 3 is that the /usr/bin/python binary is python 2 *only*, as per the pep:
For the time being, all distributions should ensure that python refers to the same target as python2 .

Thus, if you are using only python, then you are only building for python2.

jhdulaney, 2015-11-13:
Also from pep 394:

In order to tolerate differences across platforms, all new code that needs to invoke the Python interpreter should not specify python , but rather should specify either python2 or python3 (or the more specific python2.x and python3.x versions; see the Migration Notes ). This distinction should be made in shebangs, when invoking from a shell script, when invoking via the system() call, or when invoking in any other context.

jhdulaney, 2015-11-13:
Once again, you either need to specify that you only support python2, or you need to fix your build scripts.

dhylands, 2015-11-13:
With the noted exception (which is what we're doing).

According to your reading of the spec (which seems to ignore the exception I quoted), it's not possible to write a script which runs under both python2 and python3, since you can only specify a single program in the shebang.

The exception I quoted says that if your script supports both (all of our scripts do), then you should use python. And that's what we do.

jhdulaney, 2015-11-13:
And, I am pointing out that it should not break if python3 is the only python on the system, which it does.  If it breaks when python3 is the only python on the system, then you either do not support python3 or you have a bug.  Which is it?

dhylands, 2015-11-13:
If you'd like to show me the shebang line that I should use that will work in all 3 scenarios, then I'd be more than happy to fix things. But as far as I can tell, the ONLY shebang line that will work in all 3 scenarios is exactly:
```
/usr/bin/env python
```

jhdulaney, 2015-11-13:
It seems to me you're calling the python scripts during build externally.  You can specify whatever you want the binary to be at that point, and with just a little logic you can detect what is available.

dpgeorge, 2015-11-13:
The core build scripts (including makeqstrdata.py) run under both Python 2 and 3.

We fully comply with pep 394 (notably "If these conventions are adhered to, it will become the case that the python command is only executed in an interactive manner as a user convenience, or to run scripts that are source compatible with both Python 2 and Python 3.").

It therefore seems to my that your system/distribution is not pep 394 compliant, so best to report the problem to them.

Alternatively, just build using:
```
$ make PYTHON=python3
```

dpgeorge, 2015-11-13:
Reading through the discussion on python-dev about pep 394, it seems that the pep itself is slightly inconsistent and confused about the point at hand.

Guido proclaims https://mail.python.org/pipermail/python-dev/2014-September/136389.html that: "python" should always be the same as "python2"

And Nick Coghlan, who updated the pep, seems to agree https://mail.python.org/pipermail/python-dev/2014-September/136527.html that: when there's only python3 installed, there should be no /usr/bin/python

(And we know from Arch Linux that installing python3 only will actually give you /usr/bin/python.)

That makes it really hard to support both py2 and py3 in the same script (as @dhylands says, how to write such a shebang line?).  Maybe we need to go to python-dev and ask for clarification on this point...

jhdulaney, 2015-11-13:
Considering that Nick Coghlan had a lot of input on setting up the build environment in question, I rather doubt it is incorrect.

python should, according to the pep, point to python2 and python2 only.

If python2 is not present on the system, but only python3 is, then you won't have just python.

That said, if you scripts are called externally (which, they have to be if they're called by make), then you can add a little logic to detect which python is available, and, rather than call the script directly, call $PYTHON script.py

jhdulaney, 2015-11-14:
Looks like we're good now, thanks.

jhdulaney, 2016-07-21:
So, once again, we are failing to build on a system with only python3

jhdulaney, 2016-07-21:
../tools/make-frozen.py build/frozen_upip/ > build/_frozen_upip.c
/usr/bin/env: ‘python’: No such file or directory
Makefile:277: recipe for target 'build/_frozen_upip.c' failed
make: *** [build/_frozen_upip.c] Error 127
make: *** Deleting file 'build/_frozen_upip.c'

jhdulaney, 2016-07-21:
Maybe add a way to autodetect if /usr/bin/python3 is present at build time, and, if so, default to that?

pfalcon, 2016-07-21:
Yes, just like my comment from Nov 13, 2015 says. Your distro should auto-detect if python3 is the only one, and make it default "python". Instead of of going to each and every of a million Python projects and making long talks.

jhdulaney, 2016-07-21:
That is not what the Python pep says.

dhylands, 2016-07-21:
I'm just looking at 394 https://www.python.org/dev/peps/pep-0394/ and it says "so python should be used in the shebang line only for scripts that are source compatible with both Python 2 and 3".

A little later on it says: "One exception to this is scripts that are deliberately written to be source compatible with both Python 2.x and 3.x. Such scripts may continue to use python on their shebang line without affecting their portability."

dhylands, 2016-07-21:
I found a potential solution which is put the following at the top of the python script:
```
#!/bin/sh
''''which python3 >/dev/null 2>&1 && exec python3 "$0" "$@" # '''
''''which python2 >/dev/null 2>&1 && exec python2 "$0" "$@" # '''
''''which python  >/dev/null 2>&1 && exec python  "$0" "$@" # '''
''''exec echo "Error: I can't find python anywhere"         # '''

# Continue with python code here...
```

This makes it a shell script which uses exec. The quoting causes python to ignore the lines.

There are 4 quotes at the beginning of each line with no spaces between the quotes and the command. bash treats 4 single quotes in a row as 2 empty strings. python treats the triple quotes as a triple quoted string.

jhdulaney, 2016-07-21:
In py/mkenv.inc, you have this line:
PYTHON = python

Elsewhere in the make files, PYTHON is called, thus it attempts to call /usr/bin/python.

dpgeorge, 2019-02-12:
Addressed by 5368210e36f1d05efb0ea2c09661ebc80688880b

closed 2019-02-12 by dpgeorge