micropython/micropython #1618: stmhal: We should shutdown DMA clocks when no streams are in use.

dhylands, 2015-11-13:
Having the DMA block enabled seems to consume about 7-8 mA.

I think that it makes sense to add some type of bookkeeping so that we can turn the clock off when no DMA channels are being used.

For example, with PR #1608 applied, DMA2 is enabled (0x400000):
```
>>> import stm
>>> hex(stm.mem32[stm.RCC + stm.RCC_AHB1ENR])
'0x50000f'
```
if we disable the DMA2 block by setting that bit to zero:
```
>>> stm.mem32[stm.RCC + stm.RCC_AHB1ENR] = 0x10000f
```
then I see a 7-8mA drop in current (while just idling at the REPL).

dpgeorge, 2015-11-14:
I think all uses of DMA actually wait for the DMA to finish before returning (DMA is used only to get "smooth" transfers), so in principle we could just turn off the DMA blocks directly after each use.  But probably this would lead to very slow transfers, starting up the DMA block each time.

I like the idea to have a timeout to turn off the DMA when idle (see https://github.com/micropython/micropython/pull/1608#issuecomment-156740602).

dhylands, 2015-11-16:
This is dealt with in PR #1625, so closing this issue.

closed 2015-11-16 by dhylands