micropython/micropython #1622: Data dependent math bug

peter-conalgo, 2015-11-14:
I have a CRC16 algorithm that I ported quickly from a C example. After seeing it fail a few times (but not all the time), I added some test cases and noticed that although the tests pass on desktop Python3, they fail on Micropython.

I'm running: `MicroPython v1.5 on 2015-10-21; PYBv1.0 with STM32F405RG`

I attempted to localize the issue more, but whenever I simplify the loop it goes away. The issue would seem to be caused by a 32-bit int overflow in shift-left with assign. That affects the following conditional, although it's not considering that value. I've added an assert that micropython will hit as soon as it goes off the rails, marked below as `# impossible`.

To use this example:
```python
>>> import demo
>>> demo.test_crc16w()
```

```python
#!/usr/bin/env python3 -m pytest -s
#
# Implement a specific CRC16 

def crc16w(data, starting_value = 0):
    crc_reg = starting_value
    polynom = 0x8005
    
    print("crc16w(%r)" % data)

    for counter in range(len(data)):
        shift_register = 0x01
        while shift_register != 0x100:
            data_bit = 1 if (data[counter] & shift_register) else 0
            crc_bit = (crc_reg >> 15) & 0x1

            crc_reg <<= 1

            print('sr=0x%x db=%d cb=0x%x cr=0x%x' %
                    (shift_register, data_bit, crc_bit, crc_reg))

            if (data_bit ^ crc_bit) != 0:
                assert (not data_bit) or (not crc_bit)   # impossible
                crc_reg ^= polynom

            shift_register <<= 1

    crc0 = (crc_reg & 0x00FF)
    crc1 = (crc_reg >> 8) & 0xff

    return bytes((crc0, crc1))

def test_crc16w():
        
    try:
        from binascii import a2b_hex, b2a_hex
    except ImportError:
        from binascii import hexlify as b2a_hex
        from binascii import unhexlify as a2b_hex

    import sys
    assert sys.version_info > (3,0,0)
    
    assert crc16w(a2b_hex('0411')) == a2b_hex('3343')
    assert crc16w(a2b_hex('ff')) == a2b_hex('0202')
    assert crc16w(a2b_hex('aa')) == a2b_hex('fe01')
    assert crc16w(a2b_hex('ffaa')) == a2b_hex('f183')
    assert crc16w(a2b_hex('07ccbebab2')) == a2b_hex('8598')
    assert crc16w(a2b_hex('ffaa5500')) == a2b_hex('26f4')

pfalcon, 2015-11-16:
@peter-conalgo : Thanks for this report. It's still pretty big testcase and requires pyboard. If you have interest to clarify it further, please use Unix port (feel free to build for 32-bit mode using "make MICROPY_FORCE_32BIT=1"), and try to pinpoint actual failure point (like, test hypothesis that "The issue would seem to be caused by a 32-bit int overflow in shift-left with assign.")

If not, someone will get to it eventually too of course.

peter-conalgo, 2015-11-22:
Problem does not occur in unix micropython, either with or without `MICROPY_FORCE_32BIT` flag.

`MicroPython v1.5-173-g3d6240b on 2015-11-22; darwin version`

peter-conalgo, 2015-11-22:
Shorter test case. Fails on micropython (pyboard) passes on all others.

```python
def test3():
    cr = 0x7fff8000
    sr = 1
    db = 1
    cb = (cr >> 15) & 0x01
    cr <<= 1
    assert db == 1, 'db'
    assert cb == 1, 'cb'
    assert (db ^ cb) == 0, 'xor'
```

peter-conalgo, 2015-11-22:
Slightly tighter version:

```python
def test3(cr = 0x7fff8000):
    db = 1
    cb = (cr >> 15) & 0x01
    cr <<= 1
    assert db == 1, 'db'
    assert cb == 1, 'cb'
    assert (db ^ cb) == 0, 'xor'
```

Observations
- the line with `cb` is important; cannot be a constant
- top nibble of cr is critical

```python
>>> test3(0x7fffffff)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 7, in test3
AssertionError: xor
>>> test3(0x4fffffff)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 7, in test3
AssertionError: xor
>>> test3(0x2fffffff)
>>> test3(0x1fffffff)

pfalcon, 2015-11-22:
Ok, this looks like corner case of handling small vs long ints. It's reproducible with MICROPY_FORCE_32BIT=1 (you need first to make clean when building with that flag).

dhylands, 2015-11-22:
Here's a simplfied repl session that demonstrates the problem on the pyboard:
```
>>> cr_reg = 0xbfda814c
>>> cr_bit = (cr >> 15) & 1
>>> data_bit = 1
>>> (data_bit ^ cr_bit)
0
>>> (data_bit ^ cr_bit) != 0
True
```

dhylands, 2015-11-22:
And with that in hand, this fails on unix micropython:
```
MicroPython v1.5-168-g8ee43e2-dirty on 2015-11-22; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> cr_bit = (0xffffffffffffffffff >> 15) & 1
>>> (cr_bit ^ 1)
0
>>> (cr_bit ^ 1) != 0
True
>>> 
```

pfalcon, 2015-11-22:
Fixed in b3be4710aaa53429c78d03a86e9be909dca923c0.

closed 2015-11-22 by pfalcon

dpgeorge, 2015-11-22:
Nice work guys!