micropython/micropython #1624: Unix micropython upip causes OSError: 21

njouanin, 2015-11-14:
Hi,

The following command fails with error OSError: 21, without any additional information : 

./micropython -v -m upip install logging
Installing to: /Users/nico/.micropython/lib/
Installing logging 0.4.9.6 from https://pypi.python.org/packages/source/l/logging/logging-0.4.9.6.tar.gz
Traceback (most recent call last):
  File "", line 291, in <module>
  File "", line 286, in main
  File "", line 276, in main
  File "", line 199, in install_pkg
  File "", line 98, in install_tar
  File "", line 61, in _makedirs
  File "", line 60, in _makedirs
  File "", line 56, in _makedirs
OSError: 21

Environnement : MacOS with unix micropython built from head on master branch.

pfalcon, 2015-11-14:
Status of micropython-lib (and upip is part of it) on MacOS in unknown, some folks apparently tried it, but there're no specific feedback. So, you trying it there is appreciated, but please be ready that there will be issues here and there, which I'd be glad to fix though, as much as I could w/o access to MacOS (so you'll need to do all testing on your side).

So, MacOS errno 21 is EISDIR according to http://www.opensource.apple.com/source/xnu/xnu-1456.1.26/bsd/sys/errno.h , which is good, as that's consistent with Linux (and hopefully all "standard" errno's are derived from proto-Unix).

But how I can explain the above behavior is that MacOS, when asked to create a directory which exists, throws EISDIR, whereas Linux - EEXIST (17). Please try it on pre-existing dir and report results. For me on Linux:

~~~
>>> import _os
>>> _os.mkdir("/tmp")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 17
~~~

njouanin, 2015-11-14:
same test on macos : 

  >>> import _os
  >>> _os.mkdir("/tmp")
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
  OSError: 17

pfalcon, 2015-11-14:
The result is the same as on Linux, so then https://github.com/micropython/micropython-lib/pull/53 can't be right.

njouanin, 2015-11-14:
Error 21 is returned when trying to mkdir on / : 
>>> _os.mkdir("/")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 21

pfalcon, 2015-11-14:
Thanks, update upip-0.6.2 is pushed to master.

closed 2015-11-14 by pfalcon