micropython/micropython #1630: wipy: The board doesn't respond to pings

dhylands, 2015-11-17:
Once the board is assigned an IP address, I can telnet to it and frp to it, but it doesn't respond to pings.

I think it would also be useful to be able to originate a ping request to an IP address.

danicampora, 2015-11-17:
Weird, I'll investigate...

dhylands, 2015-11-17:
Hmm. I tried it again today and it seems to be working fine. Let me try a few things to eliminate things. I was flipping back and forth between networks (wipy as AP, wifi as STA) and I may have been confused about where it really was.

dhylands, 2015-11-17:
I can't reproduce this today. Everything I've done seems to be working fine. So I'm going to write this off for now.

dhylands, 2015-11-17:
ok - I tried again, and it didn't respond to pings, then I could telnet into it and then it responded to pings. I let it go for 6 seconds before hitting Control-C the first time. Here is a log of my terminal session:
```
dhylands@dave ~ 
519 >ping 192.168.0.129
PING 192.168.0.129 (192.168.0.129) 56(84) bytes of data.
From 192.168.0.20 icmp_seq=1 Destination Host Unreachable
From 192.168.0.20 icmp_seq=2 Destination Host Unreachable
From 192.168.0.20 icmp_seq=3 Destination Host Unreachable
From 192.168.0.20 icmp_seq=4 Destination Host Unreachable
From 192.168.0.20 icmp_seq=5 Destination Host Unreachable
From 192.168.0.20 icmp_seq=6 Destination Host Unreachable
^C
--- 192.168.0.129 ping statistics ---
7 packets transmitted, 0 received, +6 errors, 100% packet loss, time 5999ms
pipe 4
dhylands@dave ~ 
520 >telnet !$
telnet 192.168.0.129
Trying 192.168.0.129...
Connected to 192.168.0.129.
Escape character is '^]'.
MicroPython v1.5-2-g126373a on 2015-11-16; WiPy with CC3200
Login as: micro
Password: 
Login succeeded!
Type "help()" for more information.

>>> 
telnet> quit
Connection closed.
dhylands@dave ~ 
521 >ping 192.168.0.129
PING 192.168.0.129 (192.168.0.129) 56(84) bytes of data.
64 bytes from 192.168.0.129: icmp_seq=1 ttl=128 time=100 ms
64 bytes from 192.168.0.129: icmp_seq=2 ttl=128 time=20.8 ms
64 bytes from 192.168.0.129: icmp_seq=3 ttl=128 time=42.8 ms
64 bytes from 192.168.0.129: icmp_seq=4 ttl=128 time=66.5 ms
64 bytes from 192.168.0.129: icmp_seq=5 ttl=128 time=86.7 ms
64 bytes from 192.168.0.129: icmp_seq=6 ttl=128 time=108 ms
64 bytes from 192.168.0.129: icmp_seq=7 ttl=128 time=29.8 ms
64 bytes from 192.168.0.129: icmp_seq=8 ttl=128 time=51.8 ms
64 bytes from 192.168.0.129: icmp_seq=9 ttl=128 time=73.4 ms
64 bytes from 192.168.0.129: icmp_seq=10 ttl=128 time=97.4 ms
^C
--- 192.168.0.129 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9016ms
rtt min/avg/max/mdev = 20.827/67.832/108.208/29.155 ms
```

pfalcon, 2015-11-17:
@dhylands - Make sure it's not your access point. I personally got used to such behavior from various random APs - you may need to ping it out in reverse direction, or ping both from AP, or something.

dhylands, 2015-11-17:
Not sure if its relevant, but I had just switched the WiPy from static IP to dynamic IP. Next time it happens, I'll run wireshark on the sending computer and see whether I get any ARP responses.

My home network is certainly not typical (I have about 50 devices on it). I have an 8 port gigabit switch that the I have my 4 ubiquiti AP's on (the switch supports PoE and that's how I power the APs). I also have another 16-port gigbit switch connected to the 8-port switch that the desktop which was originating the ping came from. Oh and in my office, I have a small 8-port switch. So the path was desktop thru wired ethernet to switch 1 to switch 2 to switch 3 to AP to WiPy. So it's entirely believable that one of the switches had stale ARP data or something, and the problem was triggered by changing the IP on the WiPy. IIRC when the problem happened last time I had just changed from static IP to dynamic AP.

Although, in order to get an IP address it needs to talk to the DHCP server, which was attached to switch 2 in the setup above.

peterhinch, 2015-11-18:
Mine is on static IP and it responds to pings before telnet.

dhylands, 2015-11-23:
I'm going to close this for now since I haven't been able to reproduce.

closed 2015-11-23 by dhylands