micropython/micropython #1631: string encode('ascii') accepts non-ASCII chars

SpotlightKid, 2015-11-19:
```
>>> s = b'j\xc3\xb6rg'.decode('utf-8')
>>> s
'j\xf6rg'
>>> s.encode('ascii')
b'j\xc3\xb6rg'
```

I would have expected an EncodingError here, like in CPyton.

(BTW, why can't I input non-ASCII chars at the micropython unix repl? Is this a configuration issue?)

pfalcon, 2015-11-19:
MicroPython supports only utf-8, any other encoding is treated as as an alias for it. There're no plans to change that (doing so would require stong argumententation).

> (BTW, why can't I input non-ASCII chars at the micropython unix repl? Is this a configuration issue?)

This warrants separate issue.

SpotlightKid, 2015-11-19:
What is that encoding I'm seeing in line 3 of my example?

So, if I need to encode a string to bytes and ensure it's ASCII-only, I need to implement this myself?

pfalcon, 2015-11-19:
Yes.

SpotlightKid, 2015-11-19:
For posterity, I've implemented it like this:

```
b = s.encode()
assert all(i < 128 for i in b), "String may only contain ASCII chars."
```

closed 2015-11-19 by SpotlightKid

pfalcon, 2015-11-19:
Easy, right? And if you want to *convert* non-ascii string to ascii, CPython won't help you either - at most what it can do is to replace any non-ascii with static filler, to do something sensible, you'll need to do it yourself anyway. All the above is zen of MicroPython.