micropython/micropython #1634: Release 1.5.1?

pfalcon, 2015-11-21:
There were quite a bunch of commits since 1.5, so would make sense to tag 1.5.1. Of course, unless we want to wait couple of weeks and release 1.6 right away ;-).

dpgeorge, 2015-11-21:
Yes, we should bump the version.  Persistent bytecode is still WIP (see #1619) so maybe 1.6 should wait for more maturity there.  Also 64-bit NaN boxing #1593 is now much improved and I would like to get that in soon since it's a nightmare to keep rebasing it :)

pfalcon, 2015-11-23:
Ok, #1593 seems to pretty big change, so I assume you want to apply it after tagging. Then float printing stuff is finally stabilized, so it's good time to do it.

dpgeorge, 2015-11-23:
Ok, so llets classify persistent bytecode as "preliminary" and go to v1.5.1 now (or at least very soon, with minimal new commits). Then NaN boxing can go in after that.

dpgeorge, 2015-11-23:
@pfalcon @danicampora, and others: unless there's something important to get in, no merging until after 1.5.1.

dpgeorge, 2015-11-23:
Done, see 4dea24e10590928310f3b03703e966c9f774871c.

closed 2015-11-23 by dpgeorge