micropython/micropython #1636: Strange behaviour with return statement in finally clause

SpotlightKid, 2015-11-21:
The following code exhibits a strange behaviour:

```
def test():
    try:
        try:
            raise RuntimeError
        except:
            raise  # or some other error
    finally:
        print("Cleaning up.")
        return

test()
```

```
$ upy testfinally.py 
Cleaning up.
Cleaning up.
```

The `finally` clause is executed twice. Does not happen with CPython.

SpotlightKid, 2015-11-21:
Both micropython and CPython swallow the exception, though.

dpgeorge, 2015-11-21:
The following is a bit of a hack, but seems to fix this issue without breaking any existing tests:
```diff
diff --git a/py/vm.c b/py/vm.c
index f9aa0f9..d942d15 100644
--- a/py/vm.c
+++ b/py/vm.c
@@ -1032,6 +1032,9 @@ unwind_jump:;
 
                 ENTRY(MP_BC_RETURN_VALUE):
                     MARK_EXC_IP_SELECTIVE();
+                    if (currently_in_except_block) {
+                        --exc_sp;
+                    }
 unwind_return:
                     while (exc_sp >= exc_stack) {
                         if (MP_TAGPTR_TAG1(exc_sp->val_sp)) {
```

pfalcon, 2015-11-21:
What about more enclosed try levels?

dpgeorge, 2015-11-21:
> What about more enclosed try levels?

I tried with
```python
def test():
    try:
        try:
            try:
                raise RuntimeError
            except:
                raise
        except:
            raise
    finally:
        print("Cleaning up.")
        return

test()
```
and this is also fixed with the above patch.  The patch is popping the try-finally's exception handler, not the try-except one.  The issue is that the finally is executed a second time when the return is executed, because the try-finally exception handler is still on the stack.

SpotlightKid, 2015-11-23:
I have a few more cases.  Each will execute the finally clause twice.

Run and quit with Control-C.:

```
def test():
    try:
        while True:
            pass
    except NonExistingError:
        pass
    finally:
        print("Cleaning up.")
        return

test()
```

Run and quit with Control-C.:

```
import time

def test():
    try:
        while True:
            time.sleep(3)
    finally:
        print("Cleaning up.")
        return

try:
    test()
except KeyboardInterrupt:
    pass
```

Just run:

(BTW, just `raise Exception` (raising a class, not an instance) is apparently not supported.

```
def test():
    try:
        raise Exception("Boom")
    finally:
        print("Cleaning up.")
        return

try:
    test()
except KeyboardInterrupt:
    pass
```

dpgeorge, 2015-11-23:
> (BTW, just raise Exception (raising a class, not an instance) is apparently not supported.

It should be.  Please provide a separate test/issue for this.

SpotlightKid, 2015-11-23:
Well, it's just the last example, with `Exception("Boom")`replace by just `Exception`:

```
def test():
    try:
        raise Exception
    finally:
        print("Cleaning up.")
        return

try:
    test()
except KeyboardInterrupt:
    pass
```

This outputs:

```
Cleaning up.
Traceback (most recent call last):
  File "testfinally5.py", line 11, in <module>
  File "testfinally5.py", line 9, in <module>
NotImplementedError: byte code not implemented
```

I can normally do `raise Exception` at the REPL, so I think this is related to this issue and I'll wait for your assessment before opening another issue.

dpgeorge, 2015-11-23:
That is indeed very strange with "raise Exception".  It actually gives me a different error:
```
micropython: ../py/vm.c:1336: mp_execute_bytecode: Assertion `exc_sp >= exc_stack' failed
```
And the error is still there with "raise Exception()".  But "raise Exception(1)" makes it go away (but still run the finally twice).

Anyway, I checked that the "hack" fix I proposed above does actually fix all test cases given so far, including the very strange behaviour with "raise Exception".

dpgeorge, 2015-12-11:
Since the proposed patch (see above) works, I suggest we merge it, but with a comment that it could perhaps be improved / made less hacky.

dpgeorge, 2015-12-24:
I found another independent case that needed fixing, and modified the patch above to also fix this case.  Now pushed in 8047340d7532ec32bc9f2d603bffc0bc9544297f.

closed 2015-12-24 by dpgeorge

dpgeorge, 2018-09-03:
The handling of return-within-finally was further improved, with more test cases, in b735208403a54774f9fd3d966f7c1a194c41870f