micropython/micropython #1650: Implement Asynchronous SPI Slave

ryannathans, 2015-11-26:
This issue is to implement asynchronous sending and receiving on the SPI when configured as a slave - instead of blocking and waiting for whenever a clock is provided from the master. SPI (read and write) should probably have a configurable buffer like uart rx currently does. When the master clocks, the contents of the send buffer should be clocked out and whatever the master is sending should be clocked in to the read buffer (if exists). The user should probably be able to choose between blocking and non-blocking modes as mentioned in #1533.

Related: #1642 (async uart tx)

Bug with existing implementation:
Currently when reading SPI as a slave, the function blocks until the specified number of bytes is received or the timeout occurs. If the timeout occurs an exception is raised (OS: 116) #1414 . An exception shouldn't be raised. Whatever data was read should be returned, just as the uart read currently does.

ryannathans, 2015-11-26:
What should a blocking write return if the timeout is hit? Probably return number of bytes written, so return zero int?

blmorris, 2015-11-26:
This doesn't directly answer your questions here, but there is a general consensus that SPI and I2C slave modes are broken in the current class designs, and should be broken out into separate classes so that the current SPI and I2C classes become master only.
Unfortunately, the discussion on this topic was buried more than halfway down this thread so it's not surprising that it gets overlooked: https://github.com/micropython/micropython/issues/1430#issuecomment-151321274
I'm writing from a phone now, so I won't open any new issues myself, but implementation of these new slave classes should probably get their own issues.

ryannathans, 2015-11-26:
@blmorris should there be a separate issue to move slave to its own class or should I change this issue to be a more general issue for a rewrite/refactor outlining requirements? Will check the thread you mentioned when I'm not writing from a phone

pfalcon, 2015-12-13:
@ryannathans : Please elaborate if you're working on implementing subject and need this ticket to track your progress or something. If not, I'd suggest to close this ticket and #1642 and post on forum instead (in this issue tracker, we'd like to concentrate on bugreports, having too many feature requests on which nobody works blurs the vision). Thanks.

ryannathans, 2015-12-15:
@pfalcon I am beginning work on this today. I will appreciate any efforts made to assist me in this task. I am starting by documenting my ideas to expand on https://github.com/micropython/micropython/wiki/Hardware-API#the-spislave-class---spi-slave-end-point

ryannathans, 2015-12-15:
@pfalcon my project is on hold until even basic async I/O is available. I am about to go overseas on holiday for a few weeks so I do not feel it appropriate to start development on this by myself now.

Adixy, 2017-09-28:
I'm testing now the SPI comms when acting as a SLAVE
Percentage of succesfull communications, after one hour test, is 7.5%. 

And that is really bad