micropython/micropython #1653: Length of UART buffer 1 byte smaller than allocated

ryannathans, 2015-11-27:
With a jumper on the hardware across uart2 rx and tx

> uart = pyb.UART(2, baudrate=115200, read_buf_len=512)
> for n in range(1000):
>    uart.write("foo")
>len(uart.readall())
>511

Is this intended or has someone not been counting properly?

dhylands, 2015-11-27:
This is a classic problem. When you use a circular buffer with get and put pointers, when the buffer is full then the get and put pointers will be equal, which is the same condition as when the buffer is empty.

One way of solving this is to add a length, but now you have atomic update issues because put - get and the number of entries need to be updated atomically.

A second way requires that the buffer be a power of 2 in length, in which case you mask the pointers rather than wrapping them.

The simplest solution is that the buffer never fills that last location.

So its all about tradeoffs. I suppose we could allocate the buffer to be 1 entry bigger than requested.

pfalcon, 2015-11-28:
> The simplest solution is that the buffer never fills that last location.

I think this is right solution - there's no need to waste 15 bytes of RAM + increase code size while looking to save 1 byte.

dpgeorge, 2015-11-30:
This was not an oversight, but I agree that it's a gotcha.  How the circular buffer is implemented is an implementation detail, and this is independent from the definition of what the parameter `read_buf_len` should actually do.

If we keep the current implementation then the documentation should say that `read_buf_len` is an approximation to the actual amount of memory allocated.  But that makes the API non-obvious, and even with documentation people are going to fall into this trap again.  So it makes sense to me to actually provide the requested buffer size.  Or at least define `read_buf_len` to be a lower bound (ie behind the scenes there might be exactly `read_buf_len` bytes, or more) and provide an implementation which conforms to the definition.

dpgeorge, 2015-12-04:
Fixed by PR referenced above.

closed 2015-12-04 by dpgeorge