micropython/micropython #1658: test failures on Beaglebone Black

PeterKenyon, 2015-11-30:
on the beaglebone black running debian I got the following errors when running make test

472 tests performed (15282 individual testcases)
446 tests passed
3 tests skipped: cmd_showbc cmd_verbose extra_coverage
26 tests failed: array_q attrtuple1 builtin_range bytes exception1 gen_yield_from gen_yield_from_close gen_yield_from_ducktype gen_yield_from_exc gen_yield_from_iter gen_yield_from_send gen_yield_from_throw generator1 generator_return lexer list_clear list_copy memoryview1 memoryview_gc string1 string_splitlines sys1 machine_mem math_fun file_with print_exception
make: *** [test] Error 1

on looking at the *.exp and *.out files I noticed that the .exp failures just contained the single line
CPYTHON3 CRASH

see http://forum.micropython.org/viewtopic.php?f=2&t=1177&start=20

pfalcon, 2015-11-30:
Please start "python3" and see what version it is.

PeterKenyon, 2015-11-30:
root@beaglebone:~# python3
Python 3.2.3 (default, Feb 20 2013, 22:32:37)
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>>

PeterKenyon, 2015-12-01:
Just noticed this:
"make" is used to build the components, or "gmake" on BSD-based systems. You will also need bash and Python (at least 2.7 or 3.3).
BBB has 2.7 and 3.2.3

dpgeorge, 2015-12-08:
We run tests against CPython 3.3+ because that's the language spec that we try to aim for.  So it's not possible or sensible to use CPython 3.2 to *run the tests*.  It can be used to build and run pyboard.py (I hope!) but not for testing.

pfalcon, 2015-12-08:
I missed to comment on this all this time. Let's change subject to "Confusing failure running testsuite against CPython 3.2 or lower" and consider how to avoid such confusion.

PeterKenyon, 2015-12-08:
OK that makes sense, off topic but I don't understand why the BBB latest image has 3.2.3 and defaults to 2.7, I have successfully built and run micropython on the BBB however.

pfalcon, 2015-12-11:
Ok, so checking CPython version is of course simple to do, but there's one caveat: there's "run-test --write-exp" feature, and after that, it's possible to run testsuite without having testsuite at all. So, check for cpython version should be smart, like not run it, if cpython itself wouldn't be run. Or alternatively, add a --no-cpython option, though of course doing it automagically is better.

dpgeorge, 2020-06-09:
Closing due to lack of activity.

closed 2020-06-09 by dpgeorge