micropython/micropython #1660: Illegal instruction

texane, 2015-11-30:
I am using micropython on a Linux/ARM platform as a userland process.

I have an illegal instruction exception when running Micropython on the following setup:
/tmp # more /proc/cpuinfo 
Processor       : ARMv7 Processor rev 10 (v7l)
processor       : 0
Features        : swp half thumb fastmult vfp edsp neon vfpv3 
Hardware        : SECO i.MX6 Quad/Dual/Dual Lite/Solo Q7 Board
/tmp # ./micropython -X emit=native ./tmp.py
illegal instruction
/tmp # cat ./tmp/py
import math
print(str(math.sin(3.1415 / 2)))

I tracked down the issue to:
py/asmarm.c, asm_arm_end_pass which does not support cache flushing instruction

I do not know how to properly fix the issue so that it does not impact other ARM
based platform, but here is the patch that makes it work:

diff --git a/py/asmarm.c b/py/asmarm.c
index 2c389ac..eb8b2b9 100644
--- a/py/asmarm.c
+++ b/py/asmarm.c
@@ -85,6 +85,7 @@ void asm_arm_start_pass(asm_arm_t *as, uint pass) {
 void asm_arm_end_pass(asm_arm_t *as) {
     if (as->pass == ASM_ARM_PASS_EMIT) {
 #ifdef __arm__
+#if 0 /* FIXME */
         // flush I- and D-cache
         asm volatile(
                 "0:"
@@ -93,6 +94,10 @@ void asm_arm_end_pass(asm_arm_t *as) {
                 "mov r0, #0\n"
                 "mcr p15, 0, r0, c7, c7, 0\n"
                 : : : "r0", "cc");
+#else
+       char* const p = (char*)as->code_base;
+       __builtin___clear_cache(p, p + as->code_size);
+#endif
 #endif
     }
 }

Cheers,

dpgeorge, 2015-11-30:
We would need to find a compiler macro to test for this feature... but most likely that doesn't exist and we'd need to check for it at runtime.

texane, 2015-11-30:
Thanks for the answer. So what do you suggest ? Maybe something like:

#define MICROPY_CACHEFLUSH_IMPL_ASM 0
#define MICROPY_CACHEFLUSH_IMPL_GCC 1
#define MICROPY_CACHEFLUSH_IMPL MICROPY_CACHEFLUSH_IMPL_ASM /* default case
*/

dpgeorge, 2015-11-30:
The first thing is to understand which CPUs have this feature and how to detect it (either at compile time or runtime).

texane, 2015-11-30:
Sorry, I do not clearly understand ... By 'this feature', you mean the __builtin___clear_cache function ? This routine is GCC dependant. It may not be present in all GCC versions. So what I suggest is to add the MICROPY_CACHEFLUSH_IMPL_xxx macros . By default, inline assembly could be generated to stay compatible with what exists (MICROPY_CACHEFLUSH_IMPL_ASM). But it may be overriden to solve my case (MICROPY_CACHEFLUSH_IMPL_GCC). By looking in others projects, it seems this particular issue is handled similarly.

dpgeorge, 2015-11-30:
> By 'this feature', you mean the __builtin___clear_cache function ?

I meant the feature of the ARM CPU, whether it can flush the cache or not (well, whether it supports the particular set of instructions we use).  The builtin function may or may not do what we want it to.

I don't know enough about this topic, and don't have an ARM CPU to test it on, so others will need to chime in.  Commit b7235b8412a7511c94b31c0391eaf5c1b35ed3a8 was where the cache flushing was added, by @Vogtinator.  It was in response te #840.

Vogtinator, 2015-11-30:
Cache flushing on ARM is a PITA. There are several possible combinations of CP15 instructions that have to be run in the correct order, but IIRC, this one is not available in USR mode.
On GCC on linux, __clear_cache should be used, __builtin_clear_cache is not enough.
As there is no generic solution, I vote for MICROPY_CACHEFLUSH_IMPL_xxx macros as well.

texane, 2015-11-30:
Thanks for the precisions. I am not a pro either. What I know is it is
correct to flush the cache on non coherent cache architectures (ARM). But
it may not valid to do it from Linux userspace because of the CPU mode, or
on an ARM CPU that does not support these instructions. I do not think it
can be checked at runtime in a generic way, but I may be wrong. The only
solution I see is to add macros, and by default use the one that covers
most cases ... I had similar issues with some DMA codes, but the code runs
in a driver (ie. Linux kernel, different CPU mode).

dpgeorge, 2019-07-25:
This should be fixed by 82dc9856b06da89463e955abe2ee73a613718430

closed 2019-07-25 by dpgeorge