micropython/micropython #1661: Assertion "py/mpprint.c: 531: mp_vprintf: Assertion `0' failed" no longer holds

texane, 2015-11-30:
I run Micropython from a C program. When retrieving mp_float_t values
with mp_obj_float_get(), I get the following error:
./micropython: ../py/mpprint.c: 531: mp_vprintf: Assertion `0' failed.

I checked the mp_format_float implementation, and it looks to me that
this assertion no longer holds ...

When removing the assertion, my code works fine.

diff --git a/py/mpprint.c b/py/mpprint.c
index 54b0e47..f751e9a 100644
--- a/py/mpprint.c
+++ b/py/mpprint.c
@@ -517,19 +517,9 @@ int mp_vprintf(const mp_print_t *print, const char *fmt, va_list args) {
             case 'g':
             case 'G':
             {
-#if MICROPY_FLOAT_IMPL == MICROPY_FLOAT_IMPL_FLOAT
+#if ((MICROPY_FLOAT_IMPL == MICROPY_FLOAT_IMPL_FLOAT) || (MICROPY_FLOAT_IMPL == MICROPY_FLOAT_IMPL_DOUBLE))
                 mp_float_t f = va_arg(args, double);
                 chrs += mp_print_float(print, f, *fmt, flags, fill, width, prec);
-#elif MICROPY_FLOAT_IMPL == MICROPY_FLOAT_IMPL_DOUBLE
-                // Currently mp_print_float uses snprintf, but snprintf
-                // itself may be implemented in terms of mp_vprintf() for
-                // some ports. So, for extra caution, this case is handled
-                // with assert below. Note that currently ports which
-                // use MICROPY_FLOAT_IMPL_DOUBLE, don't call mp_vprintf()
-                // with float format specifier at all.
-                // TODO: resolve this completely
-                assert(0);
-//#error Calling mp_print_float with double not supported from within printf

dpgeorge, 2015-11-30:
Yes, since 9aaccd4735d60798963212d72989a9b9d852fc6d the mp_format_float function can handle doubles.  So this code in mpprint.c needs to be revisited.

texane, 2015-11-30:
Fine. I submited a pull request

pfalcon, 2015-12-02:
Fixed.

closed 2015-12-02 by pfalcon