micropython/micropython #1664: ViperTypeError in binary operation if len(array) used

texane, 2015-12-01:
When executing:
```
def matmul(y, a, x, b):
    # WORKS IF n = 8 is used instead
    n = len(y)
    for i in range(0, n):
        y[i] = 0.0
        for j in range(0, n): y[i] += a[i * n + j] * x[j]
        y[i] += b[i]

y = []
a = []
x = []
b = []

for i in range(0, 8): y.append(float(0.0))
for i in range(0, 64): a.append(float(i))
for i in range(0, 8): x.append(float(0.5))
for i in range(0, 8): b.append(float(0.45))
matmul(y, a, x, b)
```

I get the following error:
```
/tmp # ./micropython_fast -X emit=viper ./matmul.py 
Traceback (most recent call last):
  File "./matmul.py", line 5, in matmul
ViperTypeError: can't do binary op between 'int' and 'object'
```

It works with emit=native|bytecode. Or if n = 8, see comments

peterhinch, 2015-12-01:
Viper is undocumented and still somewhat experimental. Sometimes it has to be given hints about the types of objects. The example works if you replace ``n = len(y)`` with ``n=int(len(y))``

If you're using Viper you presumably want performance. I'd expect arrays to be faster than lists: try the following
```python
import array
@micropython.viper
def matmul(y, a, x, b):
    n = int(len(y))
    for i in range(0, n):
        y[i] = 0.0
        for j in range(0, n):
            y[i] += a[i * n + j] * x[j]
        y[i] += b[i]

y = array.array('f', (0.0,)*8)
a = array.array('f', (float(x) for x in range(64)))
x = array.array('f', (0.5,)*8)
b = array.array('f', (0.45,)*8)
matmul(y, a, x, b)
```

texane, 2015-12-01:
Thanks for the answer and the tips on arrays. I did not try to
optimize much since at the end, the people who will write the
code will likely use 'naive' approaches. So my idea is to bench
this 'worst' case. The matmul algorithm itself could be written
differently ...

peterhinch, 2015-12-01:
Until Viper is further developed and documented I'd regard it as a feature to avoid unless essential; in other words once all other attempts at optimisation have reached a limit, and more performance is required. I've used it several times to great effect but it isn't (yet) easy.

texane, 2015-12-01:
Thanks for this advice, I will keep that in mind.

dpgeorge, 2015-12-08:
Closing because it's not a bug.  Viper needs documentation, which is already known :)

closed 2015-12-08 by dpgeorge