micropython/micropython #1669: Self calibrated ADC

chuckbook, 2015-12-02:
With the appearance of the new pyboards it might make sense to introduce ADC self calibration utilizing the calibration values provided by ST.

Running from 2xAA(A) cells with proper internal temperature and voltage measurement is doable.

If there is enough interest in adding these features to stmhal I will prepare a PR.

dpgeorge, 2015-12-04:
I'm not familiar with the self calibration stuff.  Could you elaborate?

> Running from 2xAA(A) cells with proper internal temperature and voltage measurement is doable.

I never considered running on anything less than 3.3v... but I guess you can by connecting VIN to 3V3, and connecting that to 3V from 2xAA cells.

chuckbook, 2015-12-04:
Even the pyboard runs down to ~1.9V (over temp. range, at room temp. even less). Of course the freq has to be lowered or maybe just a modified flash handling. Together with the low standby power and semi-permanent freq configuration this makes an ideal battery powered device.
Of course Vdd cannot be used as Vref for ADC. The STM32F4 family has  factory calibrated band-gap reference and a linear calibrated temp sensor. These features can be used to provide astonishing good precision for both voltage and temp. measurement over a rel. wide supply voltage range.

neilh10, 2015-12-05:
It would be an excellent technical feature. See "How to get the best ADC accuracy in STM32Fx Series" 
http://www.st.com/web/en/resource/technical/document/application_note/CD00211314.pdf

peterhinch, 2015-12-06:
@chuckbook The chip datasheet (para 2.2.36) includes the following "As the offset of the temperature sensor varies from chip to chip due to process variation, the internal temperature sensor is mainly suitable for applications that detect temperature changes instead of absolute temperatures. If an accurate temperature reading is needed, then an external temperature sensor part should be used." Am I missing something here?

chuckbook, 2015-12-06:
@peterhinch This is true for uncalibrated conversions. However, all the STM32F4 devices covered by mpy so for have individual two-point temp. calibration values available. F7 devices we used so far did not have these cal. values (although specified in the specs). But ST promised to provide these values in future prod. batches.
Of course I would not recommend to use the pyboard as a clinical thermometer (even when calibrated), but more because the size of the board... (sorry, couldn't resist)
Seriously, the only reason that impacts usability of the on-die temp. sensor is the fact that it is on-die.
Self heating is the limiting factor. But there are applications benefiting from such temp. sensors. LiPo charging monitors, thermal system health (die temp is key) and periodic temp. measurements out of deep sleep, to name a few.

peterhinch, 2015-12-07:
@chuckbook Thanks for that. A calibrated value would be very useful.

dpgeorge, 2016-12-01:
This has now been implemented thanks to contributions from @chuckbook.  See 87215a0f0480dd0324a1b9c1d3fc3a5c2806249d and e72353cc48f1ac2869458625aa0ebae7b6bfe24f.

closed 2016-12-01 by dpgeorge