micropython/micropython #1674: py: Should the heap be block aligned?

dhylands, 2015-12-02:
As part of investigating the cause of various segfault related to #1656 I noticed that gc_init block aligns the end pointer but not the start pointer.

For most architectures, malloc provides block aligned data, but on the x64 it doesn't. It will provide 16-byte aligned memory, but the block size is 32 bytes.

If the heap is required to be block aligned, then it should block align both the end and start pointers.

If the heap is not required to be block aligned then it should block align the length (end - start) downwards, and add that size to the start to arrive at the end.

In extreme (not practical) terms, you can pass in an end pointer which is greater than the start pointer and gc_init can cause a segfault because it aligns end before start and then tries to memset from start for end-start bytes (which is a really big number). A concrete example of this is passing in a start pointer of 0x2010 and an end pointer of 0x2018 (so an 8-byte heap - as I said not practical). gc_init will then block align end pointer down to 0x2000 and compute the heap size as 0x2000 - 0x2010 which is an extremely large positive number.

One potential solution was provided in #1656 but @pfalcon didn't like it and said he would simplify it, so I'm raising this issue so it doesn't get forgotten.

dpgeorge, 2015-12-02:
I can't think of/recall any reason why the GC needs to have the heap block aligned.

pfalcon, 2016-01-10:
Closing this then.

closed 2016-01-10 by pfalcon

dhylands, 2016-01-10:
I think that perhaps it makes sense for the heap blocks to be cache line aligned for those architectures which have cache lines (I.e. Linux and possibly M7)

pfalcon, 2016-01-10:
Yup, but do you hint that this ticket should be open? Maintainers of affected ports are welcome to ensure (mostly automagic on POSIXish OSes for example).

dhylands, 2016-01-11:
I think its fine to leave this closed. We can post a new PR if desired. 

In userspace, linux doesn't appear to make any effort to make allocations cache line aligned.

For example, I was able to allocate 32 bytes and got back an address of: 0x1617070 which makes the 32 byte object straddle 2 cache lines (/proc/cpuinfo reports cache_alignment of 64).

For the M7, it appears to only have 8-byte (64bit) cache lines, so I think everything is fine for stmhal.