micropython/micropython #1675: Updating firmware broke pin.high() and pin.low()

ryannathans, 2015-12-03:
<Stack trace here>
AttributeError: 'bound_method' object has no attribute 'low'

offending line is pin.low()

will post more information when I get home - on mobile device now. Flashing micropython from last month resolves the problem, reflashing latest micropython brings the issue back.

dhylands, 2015-12-03:
Can you print the pin object? The fact that it says its a bound method suggests to me that it isn't really a pin object.

ryannathans, 2015-12-03:
Yeah. It's definitely a pin. The code works on micropython from mid october without any issue, flashing today's git pull with my unchanged, previously working code fails. I don't have a device with me right now to test things with but I can write up an approximation of my code. I will update my original post.

ryannathans, 2015-12-03:
It's definitely a pin. I've updated my original post.

dhylands, 2015-12-03:
I modified your code enough so that I could run it. The problem seems to be something related to class members. When the class member is defined, the sleep_pin is a pin object. But once we have an instance of MyObject accessing the class members through the instance pointer returns a bound method, but accessing the class members through the class works.

I simplified your example down to this:
```
import pyb

class MyObject:
    sleep_pin = pyb.Pin("C2", mode=pyb.Pin.OUT_PP)
    print('sleep_pin =', sleep_pin)

    def _reset_registers(self, register):
        MyObject.mode_pin.high()
        self.sleep_pin.high()

x = MyObject()
print('MyObject.sleep_pin =', MyObject.sleep_pin)
print('x.sleep_pin =', x.mode_pin)
x._reset_registers(1)
```
running the above, I get:
```
MicroPython v1.5.1-58-gb4eccfd on 2015-12-03; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> import pin_test
sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
MyObject.sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
x.sleep_pin = <bound_method>
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pin_test.py", line 14, in <module>
  File "pin_test.py", line 9, in _reset_registers
AttributeError: 'bound_method' object has no attribute 'high'
```

You didn't provide the version of the firmware that failed.

I poked around (did a bisection) and was able to narrow it down to 2f96b1982a84b21788756d9220b5c4ead657e8f7 as the commit that causes the problem to start showing up.

That commit fails, and the commit prior works:
```
MicroPython v1.4.6-123-gbedab23 on 2015-12-03; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> import pin_test
sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
MyObject.sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
x.sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
```

peterhinch, 2015-12-03:
There's surely something adrift here. The code sample has
```python
print('x.sleep_pin =', x.mode_pin)
```
yet the class has no member ``mode_pin``. I get
```python
sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
MyObject.sleep_pin = Pin(Pin.cpu.C2, mode=Pin.OUT_PP)
Traceback (most recent call last):
  File "<stdin>", line 13, in <module>
AttributeError: 'MyObject' object has no attribute 'mode_pin'
```

dhylands, 2015-12-03:
sleep_pin was assigned as part of the class, so it should be a class attribute, and should accessible through the instance or the class. In CPython:
```
class Pin:
    def high(self):
        print('pin.high')

class MyObject:
    sleep_pin = Pin()

    def _reset_registers(self, register):
        MyObject.sleep_pin.high()
        self.sleep_pin.high()

x = MyObject()
print('MyObject.sleep_pin =', MyObject.sleep_pin)
print('x.sleep_pin =', x.sleep_pin)
x._reset_registers(1)
```
produces:
```
MyObject.sleep_pin = <__main__.Pin object at 0x7fbb4fa55c50>
x.sleep_pin = <__main__.Pin object at 0x7fbb4fa55c50>
pin.high
pin.high
```
and running that python file on the pyboard also works:
```
>>> import pin_test2
MyObject.sleep_pin = <Pin object at 20003410>
x.sleep_pin = <Pin object at 20003410>
pin.high
pin.high
```
on both the old and new firmware. So its definitely something that changed in the pyb.Pin class itself.

dhylands, 2015-12-03:
The problem seems to be related to the .call method. If I comment it out:
```
const mp_obj_type_t pin_type = {
    { &mp_type_type },
    .name = MP_QSTR_Pin,
    .print = pin_print,
    .make_new = pin_make_new,
    //.call = pin_call,
    .locals_dict = (mp_obj_t)&pin_locals_dict,
};
```
then everything works as before.

I thought I would try adding ```__call__``` method to the CPython test, and that works fine:
```
class Pin:
    def __call__(self, val=None):
        if val is None:
            return 42
        else:
            print('pin set to', val)

    def high(self):
        print('pin.high')

class MyObject:
    sleep_pin = Pin()

    def _reset_registers(self, register):
        MyObject.sleep_pin.high()
        self.sleep_pin.high()

x = MyObject()
print('MyObject.sleep_pin =', MyObject.sleep_pin)
print('x.sleep_pin =', x.sleep_pin)
MyObject.sleep_pin(3)
x.sleep_pin(3)
```
giving this under CPython:
```
507 >python3 pin_test2.py 
MyObject.sleep_pin = <__main__.Pin object at 0x7f1d42b2f9b0>
x.sleep_pin = <__main__.Pin object at 0x7f1d42b2f9b0>
pin set to 3
pin set to 3
```
and this under MicroPython
```
>>> import pin_test2
MyObject.sleep_pin = <Pin object at 20003750>
x.sleep_pin = <bound_method>
pin set to 3
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pin_test2.py", line 22, in <module>
TypeError: function takes 2 positional arguments but 3 were given
```
So I don't think that the Pin class is at fault. It seems to be something to do with how class stuff is looked up/resolved.

dhylands, 2015-12-03:
The unix version of micropython prints some slightly more verbose output, which may be relevant:
```
512 >./micropython ../../test/pin_test2.py 
MyObject.sleep_pin = <Pin object at 191ef420>
x.sleep_pin = <bound_method 191ef640 <MyObject object at 191ef500>.<Pin object at 191ef420>>
pin set to 3
Traceback (most recent call last):
  File "../../test/pin_test2.py", line 22, in <module>
TypeError: __call__() takes 2 positional arguments but 3 were given
```

danicampora, 2015-12-03:
Umm very interesting.

pfalcon, 2015-12-03:
I don't think it's the same issue, but somewhat related: https://github.com/micropython/micropython/issues/1326

ryannathans, 2015-12-03:
Should title be changed if this is not limited to the Pin class? If so, what is a one line summary of the issue? I'll have a device later today to investigate further with

dhylands, 2015-12-03:
It seems to affect class members where the type of the member has a __call__ method, either in python like the example above, or natively, like the Pin class.

The runtime is incorrectly classifying the member as a bound method, when it should be just a plain value.

ryannathans, 2015-12-05:
Writing this for my own sanity...

A class member that is an instance of an object with a __call__ method incorrectly resolves as a bound_method.

A class member that is a class with a __call__ method resolves correctly.

An attribute of an instance of a class where the attribute is an instance of an object with a __call__ method resolves correctly.

Directly calling an instance of a class that has a __call__ method works correctly, instance is an object.

Directly calling an attribute of an object where the attribute is an instance of a class that has a __call__ method works correctly.

Calling a class member that is an instance of a class that has a __call__ does not work correctly and displays the error TypeError: function takes 2 positional arguments but 3 were given

Conclusion: where class members are instances of an object and that object has a __call__ method, the class member resolves incorrectly as a bound method. Additionally, the call does not work in these cases. See above TypeError.

ryannathans, 2015-12-05:
in py/runtime.c replacing line 944 with the following fixes the problem but seems to break automatic passing of "self" when creating objects
```c
        if (self == MP_OBJ_NULL && mp_obj_get_type(member) == &mp_type_fun_builtin) {
```

ryannathans, 2015-12-16:
https://github.com/micropython/micropython/blame/b8cfb0d7b2c91ea1e17f6caeabfc9cb23166cdc1/py/objtype.c#L506

this block of code seems related

If a class attribute has a __get__ method, only class.var resolves correctly, not instance.var

dpgeorge, 2015-12-16:
Thanks @ryannathans for the detailed report.  I'll get to it when possible :)

ryannathans, 2015-12-16:
@dpgeorge cheers, I spent a few days on it with a debugger. It was very easy to fix the bug manually when I needed to (setting dest[1] to null/0) but couldn't identify a way to treat class attributes in an instance like normal class attributes so no code fixes from me here :(

dpgeorge, 2015-12-26:
Should now be fixed, see link to commit above.

closed 2015-12-26 by dpgeorge