micropython/micropython #1679: WLAN constructor appears to break connection

holdenweb, 2015-12-03:
According to current documentation the `WLAN()` constructor should return the current network object. In my Wipy device, however, it invariably appears to disconnect the telnet session, which is then unrecoverable until the next hard reset.

danicampora, 2015-12-03:
It should work. What firmware version are you running?

import os
os.uname()

steveholden, 2015-12-12:
Sorry about the delay. New job ...
```
>>> os.uname()
(sysname='WiPy', nodename='WiPy', release='1.0.0', version='v1.4.6-21-gff736d6 on 2015-09-27', machine='WiPy with CC3200')
```

I've built a boot.py that seems to work to me, which you can find at https://gist.github.com/steveholden/880bdb189beaac4ca38b and I would be interested in your comments

danicampora, 2015-12-12:
You are running an old firmware. Update to the latest:

http://micropython.org/resources/firmware/wipy-2015-12-12-v1.5.1-108-gbb7f5b5.zip

steveholden, 2015-12-12:
Thanks. Obviously I haven't been following closely enough :-)  Will report back to close (I hope) in due course.

pfalcon, 2016-03-24:
Assuming fixed.

closed 2016-03-24 by pfalcon