micropython/micropython #1681: stmhal: ADC class only uses ADC1 hardware

blmorris, 2015-12-04:
This issue appears not to be very minor for the pyboard or any board using 64- or 100-pin versions of the STM32F405, so it could be low-priority or 'won't fix'. However, I encountered this running MicroPython on an Olimex STM32-E407 dev board, which uses a 144-pin STM32F407.  On this board, much of the GPIO from ports A, B, and C are connected to / utilized by on-board peripherals (like an Ethernet PHY). Most of the ADC input pins that are broken out to headers are on Port F, but those ADC pins are only connected internally to ADC3, so not accessible via the current ADC class:
```
>>> pf6 = pyb.ADC('PF6')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: pin F6 does not have ADC capabilities
>>> 
```
Since I am already customizing the firmware a bit to get it running on the Olimex board, I think that I can solve my immediate problem by hacking a few lines in `adc.c` to use ADC3 instead of ADC1. I don't have a more generic solution yet, so I just wanted to raise the issue before I forgot about it.

The biggest potential benefit that I could see in enabling the ADC2 and ADC3 hardware for lower pin-count chips would be to enable high-speed interleaved ADC reads on those pins connected to all three instances, but it would take some serious recoding to allow this, for an application that may see relatively little use.

dhylands, 2015-12-04:
I added support for all of the ADCs in the pins.csv file, so the information about which ADCs are available on which pins should be present.

To implement this without changing the existing API, I would probably reserve 5 bits to specify the channel, and then create constants ADC1 = 0, ADC2 = 32, and ADC3 = 64 which you would add to the channel number to get the channel to pass to functions like ADCAll.read_channel.

And then I would change the prototype of adc = pyb.ADC(pin) to be pyb.ADC(pin, adc=None) and have None default to ADC1.

One of the things on my todo list is to add support for ADC Sequences. This would be a list of channels that would be read on a continuous ongoing basis and would put their results in a preallocated array, and optionally support sample averaging (or something like that).

dpgeorge, 2015-12-04:
We should support ADC2 and ADC3.  But it is low priority :)

dpgeorge, 2018-11-21:
> Is there any workaround I can do to get to both ADC1 and ADC3 banks in MicroPython?

Not easily, at least not in C.  You can instead try to do it from pure Python code using the following:
```python
from micropython import const
import stm

_RCC_APB2ENR_ADC1EN_Pos = const(8)
_ADC_CR2_ADON_Pos = const(0)
_ADC_CR2_SWSTART_Pos = const(30)
_ADC_SR_EOC_Pos = const(1)

class ADC:
    def __init__(self, adc_block):
        self.adc_block = adc_block
        self.adc = {1: stm.ADC1, 2: stm.ADC2, 3: stm.ADC3}[adc_block]

    def init(self):
        # turn on CLK
        stm.mem32[stm.RCC + stm.RCC_APB2ENR] |= 1 << (_RCC_APB2ENR_ADC1EN_Pos + self.adc_block - 1)

        # reset ADC state
        stm.mem32[self.adc + stm.ADC_CR1] = 0
        stm.mem32[self.adc + stm.ADC_CR2] = 0

        # turn on ADC
        stm.mem32[self.adc + stm.ADC_CR2] |= 1 << _ADC_CR2_ADON_Pos

    def deinit(self):
        # turn off ADC
        stm.mem32[self.adc + stm.ADC_CR2] = 0

        # turn off CLK
        stm.mem32[stm.RCC + stm.RCC_APB2ENR] &= ~(1 << (_RCC_APB2ENR_ADC1EN_Pos + self.adc_block - 1))

    def read_channel(self, channel, sample_time=1):
        # select channel
        stm.mem32[self.adc + stm.ADC_SQR1] = 0
        stm.mem32[self.adc + stm.ADC_SQR2] = 0
        stm.mem32[self.adc + stm.ADC_SQR3] = channel

        # select sample time
        if channel <= 9:
            ch_pos = 3 * channel
            reg = stm.ADC_SMPR2
        else:
            ch_pos = 3 * (channel - 10)
            reg = stm.ADC_SMPR1
        reg += self.adc
        stm.mem32[reg] = (stm.mem32[reg] & ~(7 << ch_pos)) | sample_time << ch_pos

        # start conversion
        stm.mem32[self.adc + stm.ADC_CR2] |= 1 << _ADC_CR2_SWSTART_Pos

        # wait for end of conversion
        while not stm.mem32[self.adc + stm.ADC_SR] & (1 << _ADC_SR_EOC_Pos):
            pass

        # read and return value
        return stm.mem32[self.adc + stm.ADC_DR]

# test code
import time, machine
adc3 = ADC(3)
adc3.init()
pf8 = machine.Pin('F8', machine.Pin.ANALOG)
print(pf8)
for i in range(20):
    print(adc3.read_channel(6))
    time.sleep(0.1)
adc3.deinit()
```

seanlano, 2018-11-21:
Thanks heaps @dpgeorge! This seems to do just what I needed - haven't extensively tested yet but looks like it works as expected.

dpgeorge, 2022-03-30:
Additional ADCs are supported in `pyb.ADC` on H7, see ff7be31f264c359b6f11fcc2cec8489dfe4905bd