micropython/micropython #1682: stmhal: Weird test case interactions

dhylands, 2015-12-05:
As part of testing #1680 I ran into some weird test case stuff:
```
549 >./run-tests --target pyboard pyb/can.py pyb/pyb1.py pyb/modtime.py pyb/pyb1.py
pass  pyb/can.py
pass  pyb/pyb1.py
pass  pyb/modtime.py
FAIL  pyb/pyb1.py
4 tests performed (118 individual testcases)
3 tests passed
1 tests failed: pyb1
```
I had previously copied pyb1.py to /flash, if I then enter the REPL after running the above and try to import pyb1, then I get this output:
```
>>> import pyb1
3
3
(168000000, 168000000, 42000000, 84000000)
True
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyb1.py", line 38, in <module>
TypeError: 'int' object is not callable
```

dhylands, 2015-12-05:
As a followon, can & pyb1, or modtime & pyb1 works fine. It only fails in the weird way if I run can, followed by modtime followed by pyb1.

dpgeorge, 2015-12-08:
I cannot reproduce.  I have `MicroPython v1.5.1-83-g17bd241 on 2015-12-07; PYBv1.0 with STM32F405RG`.  I put pyb1.py on the flash.  I can successfully run the following:
```
$ ./run-tests --target pyboard pyb/can.py pyb/pyb1.py pyb/modtime.py pyb/pyb1.py
pass  pyb/can.py
pass  pyb/pyb1.py
pass  pyb/modtime.py
pass  pyb/pyb1.py
4 tests performed (118 individual testcases)
4 tests passed
$ ./run-tests --target pyboard pyb/can.py pyb/modtime.py pyb/pyb1.py
pass  pyb/can.py                          
pass  pyb/modtime.py
pass  pyb/pyb1.py
3 tests performed (113 individual testcases)
3 tests passed
$ picocom /dev/ttyACM0
>>> import pyb1
3
3
(168000000, 168000000, 42000000, 84000000)
True
12
```

pfalcon, 2016-01-10:
@dhylands : Do you still experience this? I never saw this, similar to @dpgeorge .

dhylands, 2016-01-11:
I tried again today and it doesn't seem to be happening anymore, so I'll close this for the time being.

closed 2016-01-11 by dhylands