micropython/micropython #1683: stmhal: tests/pyb/halerror.py leaves i2c in a weird state.

dhylands, 2015-12-05:
If I run test/pyb/halerror.py twice in a row, it fails the second time.
```
552 >./run-tests --target pyboard pyb/halerror.py pyb/halerror.py 
pass  pyb/halerror.py
FAIL  pyb/halerror.py
2 tests performed (4 individual testcases)
1 tests passed
1 tests failed: halerror
```
Examining halerror.py.out after the 2nd failure shows:
```
OSError(116,)
OSError(116,)
```
and halerror.py.exp shows:
```
OSError(5,)
OSError(116,)
```

dpgeorge, 2015-12-08:
I can reproduce this.  The I2C "recv" call is first raising a timeout error, and the second time it raises a general error, I assume because the I2C bus was left in an invalid state because the previous call failed.

The proper fix is to define more precisely how the I2C class behaves wrt errors, and then implement it.  Eg, constructing an I2C bus object should always reset the bus to a working state.

PeterKenyon, 2015-12-08:
I'm wondering if this is related to the issue I saw when a failure is only detected on the next call to the i2c device
http://forum.micropython.org/viewtopic.php?f=2&t=1027

dpgeorge, 2016-01-30:
Fixed by ea040a4f9ace1c50abc22ab755415305fd2bda41.

closed 2016-01-30 by dpgeorge