micropython/micropython #1684: `not obj` raises TypeError when __bool__ is missing.

mbuesch, 2015-12-05:
CPython 3.4.3:
```
>>> class A: pass
... 
>>> a = A()
>>> not a
False
```

`not obj` basically is equal to `obj is None` for objects not implementing `__bool__`.
But MicroPython doesn't seem to do this:

MicroPython:
```
>>> class A: pass
... 
>>> a = A()
>>> not a
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __bool__: 'A'
```

Is that difference expected? It seems rather inconsistent.

pfalcon, 2015-12-05:
This is definitely a bug.

dpgeorge, 2015-12-07:
It requires a non-trivial fix.

At the moment, "not x" is implemented as not(bool(x)), so that's why it's failing.  The reason it's implemented this way is because then a (native) type only needs to provide the "bool" unary op, and not both "bool" and "not" (which would just be opposites of each other).

We can't just provide a default "bool" for user types (or types that don't provide `__bool__`) because that would make x.__bool__() succeed for these types when it should fail.

CPython has these semantics:

- `bool(x)` always succeeds, has a sensible default
- `not x` always succeeds, has a sensible default
- `x.__bool__()` only succeeds if the type provides this method explicitly

So we need to go out of our way to conform to this.  Unless we don't care about the third point and just provide a default `__bool__`.  That would be a simple fix.

I think the only way to conform properly to CPython it is make "not" a first class unary op, but then optimise it away in mp_unary_up with a special case.

dpgeorge, 2015-12-08:
Sorry, I should have said that "not x" is currently implemented as "not x.__bool__()".

dpgeorge, 2015-12-08:
See #1692.

commit bdbe8c9a (2015-12-10): py: Make UNARY_OP_NOT a first-class op, to agree with Py not semantics.

Fixes #1684 and makes "not" match Python semantics.  The code is also
simplified (the separate MP_BC_NOT opcode is removed) and the patch saves
68 bytes for bare-arm/ and 52 bytes for minimal/.

Previously "not x" was implemented as !mp_unary_op(x, MP_UNARY_OP_BOOL),
so any given object only needs to implement MP_UNARY_OP_BOOL (and the VM
had a special opcode to do the ! bit).

With this patch "not x" is implemented as mp_unary_op(x, MP_UNARY_OP_NOT),
but this operation is caught at the start of mp_unary_op and dispatched [... cut]
closed 2015-12-10 by dpgeorge