micropython/micropython #1685: Exception.__init__ raises TypeError if overridden and called by subclass

mbuesch, 2015-12-05:
```
>>> class A(Exception):
...     def __init__(self):
...         Exception.__init__(self)
... 
>>> a = A()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 3, in __init__
TypeError: argument should be a 'Exception' not a 'A'
```

Why is `Exception` being enforced here, even if `A` is a subclass of `Exception`? That makes it hard to subclass from Exception and override `__init__`.

dhylands, 2015-12-06:
I think that the error is a bit over "agressive". I think that as you suggested it should check that self is an Exception or is derived from Exception.

As a workaround, you can recode it like:
```
class A(Exception):
    def __init__(self):
        super(A, self).__init__(self)
```

mbuesch, 2015-12-06:
Thanks, that workaround works.

peterhinch, 2015-12-07:
Or (given we're using Python3) simplify further:
```python
class A(Exception):
    def __init__(self):
        super().__init__(self)
```

mbuesch, 2015-12-07:
Out of curiosity: I'm wondering why it works with super() and does not work with direct call. Is this a MicroPython implementation detail, or are the passed types different in general?

peterhinch, 2015-12-09:
@dhylands indicated what's going on. MicroPython's support for subclassing built-in types is incomplete. See https://github.com/micropython/micropython/wiki/Differences - point 3 under "Implementation Differences".

mbuesch, 2015-12-09:
@peterhinch 
yes, I fully understand that µPy can never be fully compatible with CPython. However, It is in a pretty good shape right now and it can run quite some amount of normal Python code already. With the changes and issues I reported recently and some quirks added to my code locally, I was able to run the core of my application http://bues.ch/h/awlsim on µPy. That roughly is about 20 kLOC.
In addition to that I was able to find some bugs in my code, too, due to porting it to the new interpreter. :)

pfalcon, 2015-12-09:
So, the fact that there's incomplete implementation of Python object model in uPy is well know. Here're open bugs related to that: https://github.com/micropython/micropython/issues/606 , https://github.com/micropython/micropython/issues/624 . Some older bugs on various issues were closed on timeout as "seems to work", and may be relevant too. Few generic suggestions I have on that matter:

1. Any OO system ends up having have deep dirty roots. Don't go that deep, don't use Python-specific idiosyncrasies. Use just generic OO model.
2. If there's a workaround for your issue, use it. (Like this ticket shows.)
3. Interested parties should start looking into OO code in uPy to spot issues, find solutions, and generally refactor it to be more clear. So far, that area was driven only by me, and I'm kinda sick of it ;-).

pfalcon, 2015-12-09:
@mbuesch , We have a special topic on the forum related to that: http://forum.micropython.org/viewtopic.php?f=15&t=104 , and you're very welcome to share your experiences. That may be a good motivation for other folks to look into that or continue previous efforts. Then maybe we'll eventually get in sync and will be able to have "sprints" to progress that direction quickly ;-).

dpgeorge, 2015-12-10:
> With the changes and issues I reported recently and some quirks added to my code locally, I was able to run the core of my application http://bues.ch/h/awlsim on µPy. That roughly is about 20 kLOC.

@mbuesch that's really great to hear, and thanks very much for pushing to close a few more of the small details/holes in uPy!

jonnor, 2024-09-29:
Interestingly, on recent MicroPython v1.24.0-preview.276.g1897fe6227 (Unix port), the first code sample now fails with
```
AttributeError: type object 'Exception' has no attribute '__init__'
```

When testing the following code, based on the workaround proposed above, just specifying a message string (which seems to be mandatory on CPython).
```
class A(Exception):
    def __init__(self):
        super().__init__(self, 'msg')

raise A
```

On CPython 3.12 I get 

```
  File "/home/jon/temp/micropython/exp3.py", line 5, in <module>
    raise A
A: (A(...), 'msg')
```

On MicroPython v1.24.0-preview.276.g1897fe6227 (Unix port), there is seemingly some infinite recursion and a crash.

```
Traceback (most recent call last):
  File "exp3.py", line 5, in <module>
A: (A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(A(FATAL: uncaught NLR 0x73cfc6e06580
```